RPi-Distro/repo #1: Failure to clean up during upgrade of cgroups-bin breaks systemd

vasyugan, 2015-05-19:
When upgrading from wheezy to jessie, the file /etc/init.d/cgroup-bin is not cleaned up. 

This script tries to mount /sys/fs/cgroup/memory but seems to fail and it then unmounts /sys/fs/cgroup.
Therefore systemd cannot create new cgroups or spawning processes cannot attach to them, which breaks literally everything.

The file is not shipped by Debian, so it is a raspberrypi.org issue, see https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=785484#76

igorklopov, 2015-05-26:
@XECDesign xecdesign.com seems to be down. Please restart it. We all need it =)

XECDesign, 2015-05-26:
Not really related to this repo, but yeah, I'll get around to it eventually.

XECDesign, 2015-06-12:
It seems like you can just remove them without any problems. apt-get purge cgmanager cgroup-bin cgroup-tools libcgroup1

plugwash, 2015-06-14:
My feeling is that while this issue was caused by a raspberry pi foundation package we need to do something about this on the raspbian side since not all users upgrading will remember to include the raspberry pi foundation jessie repo.

plugwash, 2015-06-14:
ok, i've just pushed a package that should fix this (appears to work in chroot tests) to the raspbian jessie repo. Can people please test that it works on real upgrades?

XECDesign, 2015-06-22:
@plugwash out of curiosity, how did you go about fixing this?

plugwash, 2015-06-22:
see http://debdiffs.raspbian.org/main/libc/libcgroup/libcgroup_0.41-6%2brpi1.debdiff

XECDesign, 2015-06-22:
Perfect, thanks.

closed 2015-10-04 by XECDesign