RPi-Distro/repo #3: Python 3.4 support for PiFace Control & Display 2

TomStageDK, 2015-10-13:
Hi All

I have run into an issue with the python3-lirc package, a detailed description of the error i have run into can be found here: https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=122740&p=828642

But in short the python3-lirc appears to be locked in compatibility to Python 3.3 and not install able with Python 3.4.

Cheers

Tom

vkcatfish, 2015-10-17:
Hi All,

I got the same issue trying to install/use PiFaceCAD - I missing python3-lirc.  When trying to install python3-lirc, I get the following error:

pi@raspberrypi ~ $ sudo apt-get install python3-lirc
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 python3-lirc : Depends: python3 (< 3.3) but 3.4.2-2 is to be installed
E: Unable to correct problems, you have held broken packages.

I agree with Tom's synopsis: python3-lirc is not compatible with Python 3.4+.  Can anyone fix this?

Cheers,
Vince

XECDesign, 2015-10-17:
Thanks, the maintainer has already been notified and a package rebuilt for jessie should be coming.

XECDesign, 2015-11-11:
I believe this should've been fixed a while back. Closing.

closed 2015-11-11 by XECDesign