RPi-Distro/repo #4: Segfault in gst-plugins-bad1.0 / glimagesink

gkreidl, 2015-10-19:
The default video output of gstreamer1.0-1.4.4 on Jessie produces a segfault. A simple example:

gst-launch-1.0 playbin uri="file:///media/Volume/temp/big_buck_bunny_480p.ogv"
Leitung wird auf PAUSIERT gesetzt ...
Leitung läuft vor …
Kontext von Element »glimagesink0« erhalten: gst.gl.GLDisplay=context, gst.gl.GLDisplay=(GstGLDisplay)"\(GstGLDisplayEGL\)\ gldisplayegl0";
Caught SIGSEGV
#0  0x76bcb730 in poll () at ../sysdeps/unix/syscall-template.S:81
#1  0x76cb5528 in g_main_context_poll (priority=<optimized out>, n_fds=2, 
#2  g_main_context_iterate (context=0x1cf5000, block=block@entry=1, 
#3  0x76cb58ec in g_main_loop_run (loop=0x1cf1cb0)
#4  0x76dfdffc in gst_bus_poll ()
#5  0x00013db8 in ?? ()
Spinning.  Please run 'gdb gst-launch-1.0 29106' to continue debugging, Ctrl-C to quit, or Ctrl-\ to dump core.

The same happens with autovideosink or using glimagesink directly.
fbdevsink or ximagesink do work, but hey are too slow to be useful.

In fact this means that using gstreamer1.0 to display any video on the RPi is impossible right now.

XECDesign, 2015-10-19:
Could you please provide more information on how to reproduce the issue? I am seeing slightly different output.

```
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
ERROR: from element /GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0/GstOMXTheoraDec-omxtheoradec:omxtheoradec-omxtheoradec0: GStreamer error: negotiation problem.
Additional debug info:
gstvideodecoder.c(2186): gst_video_decoder_chain (): /GstPlayBin:playbin0/GstURIDecodeBin:uridecodebin0/GstDecodeBin:decodebin0/GstOMXTheoraDec-omxtheoradec:omxtheoradec-omxtheoradec0:
decoder not initialized
ERROR: pipeline doesn't want to preroll.
Setting pipeline to NULL ...
Freeing pipeline ...
```

gkreidl, 2015-10-19:
You don't even get as far as I do because the decoding doesn't start. Do you have all gstreamer1.0 plugins installed?

XECDesign, 2015-10-19:
```
pi@serge-pi ~ $ aptitude search 'gst~i'
i A gir1.2-gst-plugins-base-0.10                                                                  - Description: GObject introspection data for the GStreamer Plugins Base library                         
i A gir1.2-gst-plugins-base-1.0                                                                   - Description: GObject introspection data for the GStreamer Plugins Base library                         
i A gir1.2-gstreamer-0.10                                                                         - Description: GObject introspection data for the GStreamer library                                      
i A gir1.2-gstreamer-1.0                                                                          - Description: GObject introspection data for the GStreamer library                                      
i A gstreamer0.10-alsa                                                                            - GStreamer plugin for ALSA                                                                              
i A gstreamer0.10-plugins-base                                                                    - GStreamer plugins from the "base" set                                                                  
i A gstreamer1.0-alsa                                                                             - GStreamer plugin for ALSA                                                                              
i A gstreamer1.0-libav                                                                            - libav plugin for GStreamer                                                                             
i A gstreamer1.0-omx                                                                              - GStreamer OpenMAX plugins                                                                              
i   gstreamer1.0-plugins-bad                                                                      - GStreamer plugins from the "bad" set                                                                   
i A gstreamer1.0-plugins-base                                                                     - GStreamer plugins from the "base" set                                                                  
i A gstreamer1.0-plugins-good                                                                     - GStreamer plugins from the "good" set                                                                  
i   gstreamer1.0-tools                                                                            - Tools for use with GStreamer                                                                           
i A gstreamer1.0-x                                                                                - GStreamer plugins for X11 and Pango                                                                    
i   libgstreamer-plugins-bad0.10-0                                                                - GStreamer development files for libraries from the "bad" set                                           
i   libgstreamer-plugins-bad0.10-dev                                                              - GStreamer development files for libraries from the "bad" set                                           
i   libgstreamer-plugins-bad1.0-0                                                                 - GStreamer development files for libraries from the "bad" set                                           
i   libgstreamer-plugins-bad1.0-dev                                                               - GStreamer development files for libraries from the "bad" set                                           
i   libgstreamer-plugins-base0.10-0                                                               - GStreamer libraries from the "base" set                                                                
i   libgstreamer-plugins-base0.10-dev                                                             - GStreamer development files for libraries from the "base" set                                          
i   libgstreamer-plugins-base1.0-0                                                                - GStreamer libraries from the "base" set                                                                
i   libgstreamer-plugins-base1.0-dev                                                              - GStreamer development files for libraries from the "base" set                                          
i A libgstreamer0.10-0                                                                            - Core GStreamer libraries and elements                                                                  
i A libgstreamer0.10-dev                                                                          - GStreamer core development files                                                                       
i   libgstreamer1.0-0                                                                             - Core GStreamer libraries and elements                                                                  
i A libgstreamer1.0-dev                                                                           - GStreamer core development files                                                                       
i A libreoffice-avmedia-backend-gstreamer                                                         - GStreamer backend for LibreOffice
```

XECDesign, 2015-10-19:
Since these packages are maintained by collabora, this and #5 have been forwarded to the relevant maintainers there.

gkreidl, 2015-10-19:
Thanks. I already received an email from Emanuele and he promised to look into #5 and I suppose into this issue as well.

closed 2017-01-23 by XECDesign