RPi-Distro/repo #7: Missing source for packages

ydirson, 2015-11-23:
Eg. http://archive.raspberrypi.org/debian/pool/main/g/gst-plugins-bad1.0/ has only an old version of the source.  And there must be a deeper problem allowing to get into that inconsistent state.

em-, 2015-12-08:
Apparently the sources haven't been uploaded to the Foundation archive. You can provisionally find them in the [deprecated Collabora archive](http://raspberrypi.collabora.com/pool/web/g/gst-plugins-bad1.0/).

ydirson, 2015-12-08:
Great, thanks much! 

----- Mail original -----

XECDesign, 2015-12-09:
Thanks, grabbed the source from the collabora repo and add it later. It seems that the package was built without the -sa flag, so when it was uploaded to our repo the original source wasn't included and the source package had to be skipped. Shouldn't happen in the future.

closed 2015-12-20 by XECDesign