RPi-Distro/repo #8: gst-plugins-bad1.0: early segfault with glimagesink

ydirson, 2015-11-23:
This problem looks quite different from #4, I can't even get a backtrace as the stack is apparently
corrupted, leading to NULL SP.

Unfortunately, it is hard to get more information, as I cannot get the source for this package (see #7).

gdb --args gst-launch-1.0 videotestsrc ! glimagesink
...
Got context from element 'glimagesink0': gst.gl.GLDisplay=context, gst.gl.GLDisplay=(GstGLDisplay)"\(GstGLDisplayEGL\)\ gldisplayegl0";
[New Thread 0x732ff460 (LWP 8538)]

Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x732ff460 (LWP 8538)]
0x00000000 in ?? ()
x00000000 in ?? ()
(gdb) info registers 
r0             0x1      1
r1             0x744feb88       1951394696
r2             0x1909   6409
r3             0x0      0
r4             0x744feb78       1951394680
r5             0x73b07800       1940944896
r6             0x1909   6409
r7             0x73b06dd8       1940942296
r8             0x76fa5430       1996117040
r9             0x76e47594       1994683796
r10            0x0      0
r11            0x76d8a88c       1993910412
r12            0x7676546c       1987466348
sp             0x732fecd0       0x732fecd0
lr             0x76732b5c       1987259228
pc             0x0      0x0
cpsr           0x80000010       -2147483632

ydirson, 2015-11-24:
After reflashing a sdcard with a brand new raspbian without any other (official Debian) APT source, and upgrading the packages, I don't get this crash any more, so it's obviously not a Gstreamer bug.
I'm rather suspecting that there is some binary incompatibility between Raspbian and Debian, which would not sound good at all.

ydirson, 2015-12-07:
The problem happens without any additional apt sources, simply by installing libgles2-mesa: glimagesink apparently has a problem wrt what to do with those 2 libs... there should most probably never be two of them used by the same process!

pi@raspberrypi ~ $ lsof -p 16347|grep -i GLES
gst-launc 16347   pi  mem    REG      179,2    29996  72658 /usr/lib/arm-linux-gnueabihf/libGLESv2.so.2.0.0
gst-launc 16347   pi  mem    REG      179,2   105151 135892 /opt/vc/lib/libGLESv2.so

ydirson, 2015-12-07:
There are configure flags to avoid this problem, see https://bugzilla.gnome.org/show_bug.cgi?id=759133#c1

ydirson, 2015-12-08:
I confirm that EGL suffers the same problem, as soon as glimagesink needsto use it.
Although arranging for libgles2-mesa not to be installed was pretty easy for me, removing
libelg1-mesa would require removing Xorg.

XECDesign, 2017-01-23:
As per https://github.com/RPi-Distro/repo/issues/43

closed 2017-01-23 by XECDesign