RPi-Distro/repo #9: Audio Device Settings on Jessie not working for USB cards

rbnpi, 2015-11-30:
On Jessie the GUI Audio Device Settings sees my 3 Audio Cards {internal bcm2835 ALSA, USB Audio device (Alsa mixer) and IQaudIODAC (Alsa mixer)} and I can click the Make Default button for any of them, but it will not remember the setting for the USB card even though it marks it as default. It works for the other two.
Alsa Mixer shows the three cards 0 bcm2835, 1 USB Audio and 2 IQaudIODAC
If I edit the .asoundrc file in the Pi home directory directly and modify it to read
pcm.!default {
type hw
card 1
}

ctl.!default {
type hw
card 1
}

then I can successfully play through the USB card.
If I then look at Audio Device settings again, it has the USB card marked as default, and if I change it to another card and set it as default it does so, but it will not then let me change back again to the USB card without editing the .asoundrc file directly.

I also want to set the USB card as default when running on my pizero, and the same problem occurs there as on my Pi2 described above.

spl237, 2015-12-01:
Thanks for the report.

On having tested with a selection of USB devices here, I think I know what is going wrong, but it is non-trivial to fix. ALSA devices have a number of "elements", each element being an audio input or output channel. When a device is selected as default, the elements are iterated and I try to identify which one is the stereo front L+R channel, but the only way to do this is to compare against a table of previously-seen identifier strings. I suspect your USB device is returning a string which is not in this table, which means the plugin can't identify the element with which to associate the volume control. I was aware that this might be a problem, but yours is the first report of an issue in the wild.

Could you please let me know which USB audio device you are using? I can then get hold of one and confirm the above hypothesis, and can then update the plugin accordingly.

rbnpi, 2015-12-01:
The card is branded SABRENT and is sold amongst others by Amazon for £5.99 The guts are by C-Media Electronics. It works fine if I manually patch my .asoundrc to select card 1
Let me know if any of the files in /proc/asound would be useful. I can only attach individual files here, but if you want the whole thing as a tar or zip, let me know how I can get it to you.
Here are a couple that might help. (renamed with .txt extensions0

[usbmixer.txt](https://github.com/RPi-Distro/repo/files/48792/usbmixer.txt)
[stream0.txt](https://github.com/RPi-Distro/repo/files/48794/stream0.txt)

spl237, 2015-12-01:
That's really helpful, thank you, particularly as one of the random USB devices I found in the office has also got C-Media internals.

I've put a fix into the volumealsa plugin which makes my device work properly - if you'd like to try it, I've put it up on https://www.dropbox.com/s/wvvdnkr4ldmrqtu/volumealsa.so.zip?dl=0

This should replace the file /usr/lib/arm-linux-gnueabif/lxpanel/plugins/volumealsa.so - I suggest backing the original up first just in case you need to recover to this one.

rbnpi, 2015-12-01:
Thanks Simon
That sort’s it. Card now working and Vol control works.

spl237, 2015-12-01:
Excellent - many thanks for testing it; I'll include that change in the next release of the image.

XECDesign, 2016-01-11:
@spl237 is this addressed by the new lxpanel release?

spl237, 2016-01-11:
Yes - it's the fix in the lxpanel volumealsa plugin.

closed 2016-01-11 by XECDesign

rbnpi, 2016-01-11:
Is it in the distribution yet?
Robin

spl237, 2016-01-11:
It'll be in the next release; it should be available on apt in the next week or so.

XECDesign, 2016-01-11:
Since it was a small change, lxpanel went straight to the repo, so you should be able to `sudo apt-get update && sudo apt-get dist-upgrade -y` already

rbnpi, 2016-01-11:
OK thanks I’ll try that.
Robin

leif81, 2016-02-26:
I think I'm still seeing the issue (unless it's a different issue.)

@XECDesign @spl237 What version of the package should have the fix? I have lxpanel `0.7.2-1+rpi4` on Jessie.

XECDesign, 2016-02-29:
@leif81 Might be the same sort of issue, but affecting different hardware. 0.7.2-1+rpi4 is the latest version, so to add a fix for your hardware, some more information might be needed.

spl237, 2016-02-29:
Yes - can you please let us know the exact make and model of the USB sound device you are using?

leif81, 2016-02-29:
It's a Microsoft Life Chat 3000 usb headset.

This is what `dmesg` says about it when I plug it in:

```
[    3.202960] usb 1-1.5: New USB device found, idVendor=045e, idProduct=070f
[    3.202973] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber
=0
[    3.202984] usb 1-1.5: Product: Microsoft LifeChat LX-3000
[    3.202993] usb 1-1.5: Manufacturer: C-Media Electronics Inc.      
[    3.258735] systemd[1]: Starting Arbitrary Executable File Formats File System Automount Point.
[    3.259350] input: C-Media Electronics Inc.       Microsoft LifeChat LX-3000 as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.5/1-1.5:1.3/0003:045E:070F.0001/input/input0
[    3.303474] systemd[1]: Set up automount Arbitrary Executable File Formats File System Automount Point.
[    3.315033] hid-generic 0003:045E:070F.0001: input,hidraw0: USB HID v1.00 Device [C-Media Electronics Inc.       Microsoft LifeChat LX-3000] on usb-3f980000.usb-1.5/input3

```

playground, 2017-07-15:
I have the same issue with USB PnP Sound device.  Can't get any sound.

speaker-test -c2

speaker-test 1.0.28

Playback device is default
Stream parameters are 48000Hz, S16_LE, 2 channels
Using 16 octaves of pink noise
Playback open error: -2,No such file or directory

pi@raspberrypi:~ $ dmesg | grep C-Media
[    2.192135] usb 1-1.2: Manufacturer: C-Media Electronics Inc.      
[    2.208264] input: C-Media Electronics Inc.       USB PnP Sound Device as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.2/0003:0D8C:013C.0001/input/input0
[    2.275659] hid-generic 0003:0D8C:013C.0001: input,hidraw0: USB HID v1.00 Device [C-Media Electronics Inc.       USB PnP Sound Device] on usb-3f980000.usb-1.2/input2

eNTi, 2018-04-18:
ah there is more:
```
Apr 18 11:20:56 raspberrypi kernel: usb 1-1.4: USB disconnect, device number 5
Apr 18 11:20:56 raspberrypi kernel: usb 1-1.4: 1:1: cannot set freq 48000 to ep 0x1
Apr 18 11:20:56 raspberrypi kernel: usb 1-1.4: 2:1: cannot set freq 48000 to ep 0x82
Apr 18 11:20:56 raspberrypi kernel: usb 1-1.4: 13:0: cannot get min/max values for control 2 (id 13)
Apr 18 11:20:56 raspberrypi kernel: usb 1-1.4: 9:0: cannot get min/max values for control 2 (id 9)
Apr 18 11:20:56 raspberrypi kernel: usb 1-1.4: 10:0: cannot get min/max values for control 2 (id 10)
Apr 18 11:20:56 raspberrypi systemd[1]: Reached target Sound Card.
Apr 18 11:20:56 raspberrypi systemd-udevd[1718]: error opening ATTR{/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4:1.0/sound/card1/controlC1/../uevent} for writing: No such file or directory
Apr 18 11:20:56 raspberrypi systemd-udevd[1718]: Process '/usr/sbin/alsactl -E HOME=/run/alsa restore ' failed with exit code 19.
Apr 18 11:20:56 raspberrypi systemd[1]: sound.target: Unit not needed anymore. Stopping.
Apr 18 11:20:56 raspberrypi systemd[1]: Stopped target Sound Card.
```
I tried to get it to work with pulseaudio but it didn't work so I removed that. The link shows my dpkg.log if that's helpful:
https://pastebin.com/JS4qDZ0B

The only think I remember that I did was that I tried the setup on and then powered off the pi, put it where it was supposed to go, plugged everything in and restarted mumble. I didn't check the audio then because it had just worked a few minutes before. I only realized that something was not working at today.

XECDesign, 2018-04-18:
Doesn't looks like anything related to the volume plugin or the kernel has changed. Does it work on the latest image? Does it still work if you update that image?

eNTi, 2018-04-18:
@XECDesign sorry, but I don't know what you mean by "latest image". It doesn't work CURRENTLY and I've got the latest dist-upgrade installed, if that's what you mean?

XECDesign, 2018-04-18:
I mean a card freshly imaged with a file from https://www.raspberrypi.org/downloads/raspbian/

The only relevant package I see in your log file is pulseaudio, but if you've removed it, it shouldn't affect anything. Since I don't know what else has been done to the system, using a fresh install is a good way to confirm there's a problem that's not a result of anything you've done. And if it is something you've done, starting with a clean image will make it easier to figure out.