RPi-Distro/repo #11: provide kernel headers for module-compilation

herzi, 2015-12-09:
*This was originally filed at https://github.com/raspberrypi/linux/issues/1220. That ticket was closed with the recommendation to file the bug here.*

As of today, kernel module developers have to recommend against running the kernel shipped with rpi-bootloader because there are no headers to compile against. See e.g. [rtl8812AU_8821AU_linux](https://github.com/abperiasamy/rtl8812AU_8821AU_linux/blob/master/README.md#compiling-for-raspberry-pi).

This is very unfortunate, as these users cannot benefit of improvements in this kernel image.

Please consider shipping a package called `linux-headers-rpi-bootloader` or `raspberry-bootloader-headers`, so we can happily compile against a more modern Linux kernel without having to build one on our own.

XECDesign, 2015-12-09:
This has been on the todo list for quite a while. It's a little tricky since the original debian kernel source package is difficult to work with and the kernel's own package building script need some modifications and doesn't produce a source package. I suppose it's more important to have something than put it off because it's tricky to do properly, so I will give it another go. Thanks for the reminder.

herzi, 2015-12-09:
Just to make sure there's no mis-understanding: I'm not talking about a full source package. I'm just requesting a package containing the headers, so I can build kernel modules that are not included in the kernel's source tree. This should be a lot less work than creating a full source package for custom kernel compilation.

XECDesign, 2015-12-09:
No misunderstanding. It's just that packages without a proper matching debian source package are generally frowned upon, so the temptation is to put it off until there's enough time to do it properly. However, it has been put off for years, so I think it's best to just get it done and not worry too much about how.

diederikdehaas, 2015-12-20:
@XECDesign Have you tried asking on the [debian kernel ML](https://lists.debian.org/debian-kernel/) ?

diederikdehaas, 2015-12-20:
@herzi You may want to take a look at https://github.com/diederikdehaas/cknow.org/blob/master/rpi/compile-kernel-module-on-raspberrypi.md (and https://github.com/diederikdehaas/rtl8812AU)

XECDesign, 2015-12-20:
@diederikdehaas I haven't, but after discussing it with plugwash, he did:
https://lists.debian.org/debian-derivatives/2014/12/msg00001.html
I've popped in to #debian-kernel  to talk with some of the people who replied afterwards, but yeah, there's just no easy way to do it properly, from what I gather. At the time, I wasn't very familiar with debian packaging in general, so maybe I am thinking it's more complicated than it is. I'll revisit doing it properly and if that still involves digging through undocumented perl scripts, I'll do it the "good enough" way. I have the day free now, so may as well get started. Will keep you posted.

diederikdehaas, 2015-12-20:
Yep, it doesn't look easy. I just thought of that option as [I asked a question there](https://github.com/diederikdehaas/rtl8812AU/commit/65ce8988b61d0b4e457c0b40aa43ea313cadc7b2) and got the perfect answer within the hour from Ben Hutchings. Someone on the forums mentioned that he tends to be quite helpful and he is a good source regarding the kernel and the debian packages of it :wink: 

Good luck :+1:

diederikdehaas, 2016-01-14:
@herzi note that the instruction in that repo are out of date. The version of the kernel/headers you get by installing `linux-image-rpi-rpfv`/`linux-headers-rpi-rpfv` is not `3.10-3` but `3.18.0-trunk`. 
Still not as new as the kernel from `raspberrypi-bootloader`, but still better/newer then `3.10-3`.

diederikdehaas, 2016-01-15:
https://packages.debian.org/experimental/linux-headers-4.4.0-rc8-versatile ... just saying :wink:

XECDesign, 2016-01-16:
That seems promising. Will take a look later. Cheers.

diederikdehaas, 2016-02-13:
It turns out the versatile thing wasn't relevant at all.

But I'm actually posting here because of the following package: 
https://packages.debian.org/sid/kernel-package
Not sure if you're already familiar with it or if it may be useful, but wanted to mention it anyway :wink:

XECDesign, 2016-05-20:
Fixed by https://github.com/RPi-Distro/firmware/commit/7f206fe9fa9baacd56bdd0e50fd8cfcfc3611f86

closed 2016-05-20 by XECDesign