RPi-Distro/repo #12: Avoid dhcpcd as it crashes GSM/UTMS dongles

callegar, 2015-12-12:
From googling it looks like GSM and UMTS usb dongles are widely used with the rpi platform.

The most recent generation of these dongles shows up on the system both as a serial porta and as a wwan ethernet port since their usb hardware uses the CDC_ether profile. The serial port is used to set up the APN and to tell the device to connect.

These devices seem to be crashed by dhcpcd. When dhcpcd rund, the dongles are constantly rebooting. This has been practically observed with the HUAWEI E173 dongle.

No other DHCP client seems to have this side effect.
I have been unable to find a combination of options preventing the issue, apart from telling dhcpcd not to manage wwan0

XECDesign, 2017-09-08:
Is this still a problem under Stretch?

s1awa, 2017-09-11:
@callegar 
can you check if dhcpcd and networking are running at the same time?
```
sudo systemctl status networking.service
sudo systemctl status dhcpcd.service
```

If so try to disable networking
`sudo systemctl disable networking`

sergiocallegari, 2017-09-17:
@XECDesign my system was *upgraded* to stretch. Possibly this can be the reason for an issue that should not be present on stretch.

@s1awa cannot make tests right now as the item is under use by others. But I think they were both active. Unclear to me how to setup the connection via UMTS dongle with networking disabled and no gui.

XECDesign, 2017-10-31:
I've got an E173. What do I need to do to reproduce the problem?I am seeing the wwan0 interface device and the CD-ROM devices show up. I am not seeing the device reboot. Does it have to actually connect to a network?

callegar, 2017-10-31:
I'll try to get back the umts dongle and re-give it a try to provide more precise instructions to reproduce the issue soon.

closed 2018-05-30 by XECDesign