RPi-Distro/repo #13: raspberry pi foundation webkit package breaks installability of

plugwash, 2015-12-26:
The raspberry pi foundation webkit packages break the dependencies of libwebkit2gtk-3.0-25. This in turn makes anjuta , devhelp and gthumb uninstallable.

root@odroidu2:/# apt-get install anjuta
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 anjuta : Depends: libdevhelp-3-2 (>= 3.12.0-2~) but it is not going to be installed
          Depends: libwebkit2gtk-3.0-25 (>= 2.0.2) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
root@odroidu2:/#

root@odroidu2:/# apt-get install anjuta libdevhelp-3-2  libwebkit2gtk-3.0-25
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libwebkit2gtk-3.0-25 : Depends: libjavascriptcoregtk-3.0-0 (= 2.4.8-2+rpi1) but 1:2.4.1-1rpi53rpi1g is to be installed
                        Recommends: geoclue-2.0 but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
root@odroidu2:/#

XECDesign, 2016-01-11:
Unfortunately, it appears that there's no obvious fix for this for the time being. Apt pinning might be a suitable workaround.

bwims, 2016-04-17:
Hi, @XECDesign,

could you explain the mechanics of Apt pinning as a workaround?  I am desperate to install Gthumb as part of the Gphoto package on my Pi 3

Thanks!

XECDesign, 2016-04-18:
@bwims This thread might be useful https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=136036#p904924

sgon00, 2016-09-07:
I met this problem when I install gthumb in my raspberry pi 3 today. When is this issue going to be fixed?

```
pi@pi03:~ $ sudo apt-get install gthumb
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 gthumb : Depends: libwebkit2gtk-3.0-25 (>= 2.0.2) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
pi@pi03:~ $ sudo apt-get install libwebkit2gtk-3.0-25
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libwebkit2gtk-3.0-25 : Depends: libjavascriptcoregtk-3.0-0 (= 2.4.9-1~deb8u1+rpi1) but 1:2.4.1-1rpi53rpi1g is to be installed
                        Recommends: geoclue-2.0 but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```

XECDesign, 2017-09-05:
Resolved in stretch

closed 2017-09-05 by XECDesign