RPi-Distro/repo #14: Driver / gpu leaks.

mistralol, 2016-01-09:
Not sure how I triggered this but I suspect it involved crashing gstreamer pipelines. Basically the omx enc / dec broke after a while with the following error.

Using this pipeline it was no longer able to encode data.

 GST_DEBUG=omx*:3 gst-launch-1.0 -v videotestsrc is-live=true ! video/x-raw, width=1280, height=720 ! omxh264enc ! h264parse ! omxh264dec ! fakesink silent=false

I would start getting this error message. From either using the encoder or decoder.

0:00:00.272602201 24803   0xe09860 ERROR                    omx gstomx.c:1730:gst_omx_port_allocate_buffers_unlocked:<omxh264enc-omxh264enc0> Failed to allocate buffer for video_encode port 200: Insufficient resources (0x80001000)

So then by rebooting.... Running the exact same command will work perfectly fine for a while. This would probably indicate some sort of resource leak in the driver / firmware

XECDesign, 2016-01-14:
@popcornmix is this most likely to be a problem with gst omx itself or could it be a firmware issue?

mistralol, 2016-01-14:
Yes it could be a firmware issue. Though after increasing the gpu / ram split it seemed to correct it self and I have not been able to reproduce it with such ease.

closed 2017-01-23 by XECDesign