RPi-Distro/repo #15: Desktop logout option kills all desktops (including virtual)

sconemad, 2016-01-11:
This was originally reported for Raspbian on launchpad: https://bugs.launchpad.net/raspbian/+bug/1418501, and still affects the latest Wheezy and Jessie versions of Raspbian.  

It's possible to have more than one LXDE desktop running, e.g. by using RealVNC's VNC Server for Raspberry Pi to create additional virtual desktops.

However, when selecting Shutdown > Logout from the desktop menu, ALL of the running desktops are shutdown, not just the current one.

This appears to be because lxde-pi-shutdown-helper is simply running "pkill lxsession" for a logout, killing all running instances of lxsession (at least for the current user).

With LXDE, the PID of the current desktop's lxsession should be available in the _LXSESSION_PID environment variable, so I would suggest changing the above line to:

kill $_LXSESSION_PID

That way only the current desktop's lxsession is killed, leaving any others running.

XECDesign, 2016-01-11:
Seems sensible.
@spl237 any reason not to do this?

spl237, 2016-01-11:
Not that I can think of - seems a good thing to do.

XECDesign, 2016-05-01:
Should've been fixed a while back.

sconemad, 2016-05-09:
Yes, working now - thanks!

spl237, 2016-09-28:
Reported by RealVNC as a problem with new image - reopening.

spl237, 2016-09-28:
Yes, seems I'd reintroduced the old behaviour with the change to the new shutdown helper - fixed for next release.

closed 2016-09-28 by spl237