RPi-Distro/repo #16: Qt5 in Raspbian is partly broken, how can we fix it?

diederikdehaas, 2016-01-15:
The topic of getting Qt5 to work properly in Raspbian has now come up in several places.

The title of this issue is directly taken from https://www.raspberrypi.org/forums/viewtopic.php?f=67&t=122948. Another forum thread is here: https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=108978. And recently the topic was raised (mostly by me) on a raspbian bug too: https://bugs.launchpad.net/raspbian/+bug/1533378

In [this comment](https://bugs.launchpad.net/raspbian/+bug/1533378/comments/10) I said the following:
> Alternatively, if it is not possible in the context of the raspbian.org repos, we can also ask whether the people from the raspberrypi.org repos are interested. And if we can present them with the solution on how to do it rather then if we request them to figure it all out by themselves, it should have a far bigger chance of getting accepted.

But before we start on the 'whole endeavor', it's maybe a good idea to ask here first whether that could be a reasonable alternative/solution or it would be a useless exercise. 

For completeness, I'll post my whole 'story' (from that comment) in a separate post below.

diederikdehaas, 2016-01-15:
> If people want something that works with pi specific libraries

Here's the thing: I _think_ that that is not needed.

What I hope/think is that a different configuration file, qmake.conf, needs to be selected, and combined with the original Debian/Raspbian sources would make for qt5 libraries which are (well) suited to the Pi.
If you haven't seen it, here's the 'analyses' I've made thus far: https://www.raspberrypi.org/forums/viewtopic.php?p=865560&sid=afb1c09e8a477a1e6129e810f67f3d32#p865560

Right now, the only 'solution' I see on the interwebs, is:
"Here's how you can compile it (all) from source yourself" and that's exactly what the [link in the OP](http://wiki.qt.io/RaspberryPi2EGLFS) tells you.
Or "Here is some (random) repo on the internet that you can use", so you don't have to compile it all yourself.

The first 'solution' is so daunting for (al)most (all) people, that they bail out before they have even started (except for the hard-core QT fan/expert) and the second one is a massive security risk in my opinion.
Consequence of this is (imo) that qt5 will not/hardly be used on the Pi. Period.
And I think that's a pity.

I'm a big fan of KDE and QT. KDE on the Pi is a no-go.
But what is possible with QT is impressive, also on the Pi. See the youtube links ([1](https://www.youtube.com/watch?v=ID9z1iNu7OI) [2](https://www.youtube.com/watch?v=nmWBGWa6-24)) in the OP of the forum thread I linked above.
Also, LXDE is busy transforming to LXQT. 
(Whether they have progressed enough to be fully functional in Stretch remains to be seen though.)
**Correction: [apparently not](http://blog.lxde.org/?p=1310). It looks like the plan is to have 2 versions, LXDE and LXQT**

I really doubt that _I_ have the knowledge to figure it out by myself as there are various pieces which I don't know whether they are important and/or how they should be 'strung together' to make it work properly on the Pi.
But if I can connect all the people who are interested in getting qt5 working on the Pi, we should have a reasonable chance of figuring it out together.
I do know that I am willing to compile the whole stack as many times as needed. And I think there's a reasonable chance that if some debian-qt (packaging) specific knowledge is needed I can ask the right people for that (lisandro and mitya57 come to mind).

I don't know if the solution, if we find it, is possible and/or allowed in the context of the raspbian.org archive.
But I do want to give it a shot. Even if it is a long shot.

Alternatively, if it is not possible in the context of the raspbian.org repos, we can also ask whether the people from the raspberrypi.org repos are interested. And if we can present them with the solution on how to do it rather then if we request them to figure it all out by themselves, it should have a far bigger chance of getting accepted.

(full 'background story' at https://bugs.launchpad.net/raspbian/+bug/1533378)

diederikdehaas, 2016-01-15:
nvm, you can't without breaking QT for everyone else.

closed 2016-01-15 by diederikdehaas

codepainters, 2016-08-08:
While this ticket is closed, do you mind giving more details about why it's not possible to solve it? 
Why can't Raspbian deliver customized Qt5 packages, compiled `eglfs` plug-in compiled for RPi (i.e. with `-device linux-rasp-pi2-g++` ? Just like packages from https://twolife.be do?

XECDesign, 2016-08-08:
The reason why Raspbian won't do it is because it needs to run on platforms other than the pi. It's just an armv6  recompile of Debian, for the most part, as is.
It could be included in the raspberrypi.org repo, but we'd need to know how it affects existing QT5 applications and whether it would conflict with Anholt's OpenGL work. We would need input from people who know QT5 and Debian very well.

codepainters, 2016-08-08:
OK, I assumed that Raspbian cares about RPi only. If not - then then problem is obvious.

gzotti, 2017-02-28:
I would not mind compiling Qt 5.4 or later on an RPi3 from sources, if there would be any configuration (QT's configure command line) known to work with (otherwise) default Raspbian (Jessie), X (Pixel), and the Anholt VC4 drivers as of Feb.2017. It always fails with eglfs issues way beyond my understanding. Also the Qt5.5 from twolife.be appears not to work with the VC4 drivers. Stellarium (and other projects) urgently need more recent Qt than 5.3.2 from 2014! Other SoC boards have 5.5 and newer.

XECDesign, 2017-02-28:
I don't think there's a chance of qt5 packages in Raspbian compiled against Anholt's driver until we switch to them.

gzotti, 2017-02-28:
Yes, that's possible. I just wonder if really nobody has found out how to bring these to work? What changed in Qt5.4 that does not compile or makes eglfs incompatible? Qt5.3.2 work with Anholt's drivers, but we will soon require newer functionality (not even GL related) introduced with Qt5.4. 
I just found Qt5.8 is out which brings something new around EGL. Something to play again for a few weekends...

diederikdehaas, 2017-03-01:
@gzotti Have you looked into `/usr/lib/x86_64-linux-gnu/qt5/mkspecs/devices/` (or sth like that) ?
Debian Sid and Stretch, so likely also available on Raspbian Stretch, have Qt 5.7.1 and it contains the following directories (among others):
```
drwxr-xr-x 2 root root 4096 jan 27 17:43 linux-rasp-pi2-g++
drwxr-xr-x 2 root root 4096 jan 27 17:43 linux-rasp-pi-g++
drwxr-xr-x 2 root root 4096 jan 27 17:43 linux-rpi3-g++
drwxr-xr-x 2 root root 4096 jan 27 17:43 linux-rpi3-vc4-g++
```

gzotti, 2017-03-01:
Ah, that's good news, thanks. I just found a few blogs describing how to bring Debian Stretch to the RPi.  Now I hope "official" Raspbian Stretch will come sometimes this year. 
The linux-rpi3-vc4-g++ device looks promising, indeed... :-)

gzotti, 2017-03-05:
Building Qt5.8 on Raspbian Jessie worked not too badly. make -j3 failed with "libpng not found", without "-j" it built overnight, and QtScript required a change in the mkspec to set Arm8 for some 3rd-party component (sorry, don't remember name of the define, writing from another PC). After that, building Stellarium went not perfectly but ok. Cmake did not resolve macros Qt::Gui_EGL and Qt::Gui_GLESv2, which required manual linking. Started from a terminal in PIXEL, Stellarium runs in a full-screen window with limited viewport (this may be some EGL limitation, but I can switch a terminal window over it), and the current version of the VC4 driver shows a bug which Anholt may have fixed in the meantime (he could not reproduce it). It seems we are getting somewhere this year, at least.

ali1234, 2017-03-10:
The main reason it is a no go today is because jessie's Qt version is not new enough to support the eglfs plugin properly. Backporting Qt 5.4+ to jessie has a high potential to break a lot of software that expects Qt 5.3, and backporting the eglfs stuff to Qt 5.3 would be a huge task.

When Raspbian goes to stretch it will be just a matter of building the Qt package with one extra config parameter to build the eglfs plugin, plus adding a dependency on the vclibs. Since it is a plugin it will have no effect if you don't use it eg because you are running X11 or Wayland. eglfs is designed to run a single embedded-style application full screen (hence fs).

In the mean time I have created a tool to automatically cross compile Qt with eglfs (and nothing else):

https://github.com/ali1234/qt5pi-builder

This will fetch the compiler, build a sysroot directly from the repos with multistrap, checkout git sources, and build them, without using qemu or physical hardware.

gzotti, 2017-03-10:
Thanks for the explanations. I think I will wait for Stretch with (hopefully) Qt5.6 or later.

diederikdehaas, 2017-03-10:
> Stretch with (hopefully) Qt5.6 or later.

Stretch will have Qt5.7.1