RPi-Distro/repo #17: Raspberry Pi Zero USB console

notro, 2016-01-16:
It would be very convenient if I could just plug my Raspian powered Zero into a computer and get a console over USB.

I propose a way to achieve that by adding a systemd service coupled with the upcoming Linux version 4.4.

To enable this USB console, the user has to make 2 changes on the readily available boot partition:
* Add to config.txt: dtoverlay=dwc2
* Add to cmdline.txt: usbcon

The first one is needed because the dwc2 USB controller driver is used to get gadget/otg support. dwc_otg doesn't support that today.
The second is needed to activate the service that loads g_serial which provides a USB serial port and puts a console on that port.

On my Windows 8 computer the port shows up as: ELMO GMAS (COM5)
I use Putty to connect to COM5 at 115200.

The 4.4 kernel is currently available using rpi-update:
```
sudo BRANCH=next rpi-update
```

This is the proposed change:

/etc/systemd/system/usbcon.service
```
[Unit]
Description=USB console on ttyGS0

ConditionKernelCommandLine=usbcon

[Service]
Type=oneshot
RemainAfterExit=yes

ExecStart=/sbin/modprobe g_serial
ExecStart=/bin/systemctl start serial-getty@ttyGS0.service

ExecStop=/bin/systemctl stop serial-getty@ttyGS0.service
ExecStop=/sbin/modprobe -r g_serial

[Install]
WantedBy=getty.target

```

Enable service:
```
~$ sudo systemctl enable usbcon
Created symlink from /etc/systemd/system/getty.target.wants/usbcon.service to /etc/systemd/system/usbcon.service.

```

After reboot:
```
~$ systemctl status usbcon
● usbcon.service - USB console on ttyGS0
   Loaded: loaded (/etc/systemd/system/usbcon.service; enabled)
   Active: inactive (dead)
           start condition failed at Sat 2016-01-16 10:50:53 UTC; 1min 2s ago
           ConditionKernelCommandLine=usbcon was not met

```

Activate service:
/boot/cmdline.txt
```
usbcon
```

Reboot:
```
~$ systemctl status usbcon
● usbcon.service - USB console on ttyGS0
   Loaded: loaded (/etc/systemd/system/usbcon.service; enabled)
   Active: active (exited) since Sat 2016-01-16 10:56:06 UTC; 4s ago
  Process: 505 ExecStart=/bin/systemctl start serial-getty@ttyGS0.service (code=exited, status=0/SUCCESS)
  Process: 453 ExecStart=/sbin/modprobe g_serial (code=exited, status=0/SUCCESS)
 Main PID: 505 (code=exited, status=0/SUCCESS)
   CGroup: /system.slice/usbcon.service

~$ systemctl status serial-getty@ttyGS0
● serial-getty@ttyGS0.service - Serial Getty on ttyGS0
   Loaded: loaded (/lib/systemd/system/serial-getty@.service; disabled)
   Active: active (running) since Sat 2016-01-16 10:56:06 UTC; 8s ago
     Docs: man:agetty(8)
           man:systemd-getty-generator(8)
           http://0pointer.de/blog/projects/serial-console.html
 Main PID: 510 (agetty)
   CGroup: /system.slice/system-serial\x2dgetty.slice/serial-getty@ttyGS0.service
           └─510 /sbin/agetty --keep-baud 115200 38400 9600 ttyGS0 vt102

~$ lsmod | grep g_serial
g_serial                3785  0

```

Manually stop:
```
~$ sudo systemctl stop usbcon

~$ systemctl status usbcon
● usbcon.service - USB console on ttyGS0
   Loaded: loaded (/etc/systemd/system/usbcon.service; enabled)
   Active: inactive (dead) since Sat 2016-01-16 10:57:24 UTC; 5s ago
  Process: 584 ExecStop=/sbin/modprobe -r g_serial (code=exited, status=0/SUCCESS)
  Process: 580 ExecStop=/bin/systemctl stop serial-getty@ttyGS0.service (code=exited, status=0/SUCCESS)
  Process: 505 ExecStart=/bin/systemctl start serial-getty@ttyGS0.service (code=exited, status=0/SUCCESS)
  Process: 453 ExecStart=/sbin/modprobe g_serial (code=exited, status=0/SUCCESS)
 Main PID: 505 (code=exited, status=0/SUCCESS)

~$ systemctl status serial-getty@ttyGS0
● serial-getty@ttyGS0.service - Serial Getty on ttyGS0
   Loaded: loaded (/lib/systemd/system/serial-getty@.service; disabled)
   Active: inactive (dead)
     Docs: man:agetty(8)
           man:systemd-getty-generator(8)
           http://0pointer.de/blog/projects/serial-console.html

~$ lsmod | grep g_serial
~$

```

Maybe loading of g_serial should be allowed to fail in case g_multi is used to provide ttyGS0:
```
ExecStart=-/sbin/modprobe g_serial
ExecStop=-/sbin/modprobe -r g_serial
```

notro, 2016-01-16:
Maybe we can get away with just changing config.txt not having to touch cmdline.txt:
```
dtoverlay=dwc2,bootargs=usbcon
```

@pelwell can we have an empty 'bootargs' property that can be set using an overlay parameter?
If so how is the labeling done?
```
	fragment@1 {
		target-path = "/chosen";
		__overlay__ {
			bootargs = "";
		};
	};

	__overrides__ {
		bootargs = <??>, "bootargs";
	};
```
(the bootargs property contains the kernel command line arguments, the overlay mechanism concatenates them together)

Current overlay: [dwc2-overlay.dts](https://github.com/raspberrypi/linux/blob/rpi-4.4.y/arch/arm/boot/dts/overlays/dwc2-overlay.dts)

pelwell, 2016-01-16:
That ought to work. You would need to put the label on the __overlay__ node, which would end up creating a phandle within it that would overwrite any existing one when applied, but that shouldn't be a problem in this case

notro, 2016-01-26:
**Raspberry Pi Zero USB network**

This is a similar service that enables USB gadget networking.
Activated by adding to the kernel commandline: usbnet

/etc/systemd/system/usbnet.service
```
[Unit]
Description=USB gadget network on usb0

ConditionKernelCommandLine=usbnet

[Service]
Type=oneshot
RemainAfterExit=yes

# Load g_ether and use smsc95xx.macaddr for host and alter it to begin with 00 for device
ExecStart=/bin/sh -c 'host_addr="$( sed "s/^.*macaddr=\([0-9A-F:]*\) .*$/\1/" /proc/cmdline )"; dev_addr=$(echo $host_addr | sed s/^B8/00/); modprobe g_ether dev_addr=$dev_addr host_addr=$host_addr'

ExecStop=/sbin/modprobe -r g_ether

[Install]
WantedBy=multi-user.target

```
(the macaddr extraction was taken from a forum post by rpdom)

Enable service:
```
~$ sudo systemctl enable usbnet
Created symlink from /etc/systemd/system/multi-user.target.wants/usbnet.service to /etc/systemd/system/usbnet.service.

```

Pi Zero using a [link-local address](https://en.wikipedia.org/wiki/Link-local_address) (no dhcp available):
```
~$ ifconfig usb0
usb0      Link encap:Ethernet  HWaddr 00:27:eb:6b:18:5a
          inet addr:169.254.214.199  Bcast:169.254.255.255  Mask:255.255.0.0
          inet6 addr: fe80::eff7:6986:e50f:9bdd/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1

~$ hostname
pizero
```

Connected Windows host (Bonjour helps with name resolution):
```
C:\Users\Noralf>ipconfig /all

Ethernet adapter Lokal tilkobling:

   Connection-specific DNS Suffix  . :
   Description . . . . . . . . . . . : USB RNDIS Adapter
   Physical Address. . . . . . . . . : B8-27-EB-6B-18-5A
   DHCP Enabled. . . . . . . . . . . : Yes
   Autoconfiguration Enabled . . . . : Yes
   Link-local IPv6 Address . . . . . : fe80::5dc:fd38:d26d:8528%17(Preferred)
   Autoconfiguration IPv4 Address. . : 169.254.133.40(Preferred)
   Subnet Mask . . . . . . . . . . . : 255.255.0.0
   Default Gateway . . . . . . . . . :
   DHCPv6 IAID . . . . . . . . . . . : 293520785
   DHCPv6 Client DUID. . . . . . . . : 00-01-00-01-1C-53-C7-D0-54-A0-50-7B-E9-63
   DNS Servers . . . . . . . . . . . : fec0:0:0:ffff::1%1
                                       fec0:0:0:ffff::2%1
                                       fec0:0:0:ffff::3%1
   NetBIOS over Tcpip. . . . . . . . : Enabled

C:\Users\Noralf>dns-sd -Q pizero.local
Timestamp     A/R Flags if Name                             T   C Rdata
17:44:21.697  Add     2 17 pizero.local.                    1   1 169.254.214.199

C:\Users\Noralf>ping -4 pizero.local

Pinging pizero.local [169.254.214.199] with 32 bytes of data:
Reply from 169.254.214.199: bytes=32 time<1ms TTL=64

```

**Internet Connection Sharing enabled on Windows 8.1**

Pi Zero gets a dhcp address:
```
~$ ifconfig usb0
usb0      Link encap:Ethernet  HWaddr 00:27:eb:6b:18:5a
          inet addr:192.168.137.192  Bcast:192.168.137.255  Mask:255.255.255.0
          inet6 addr: fe80::24b:5bfc:8a9e:b148/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1

~$ ping raspberrypi.org
PING raspberrypi.org (93.93.130.214) 56(84) bytes of data.
64 bytes from 93.93.130.214: icmp_seq=1 ttl=50 time=33.5 ms
```

Windows:
```
C:\Users\Noralf>dns-sd -Q pizero.local
Timestamp     A/R Flags if Name                             T   C Rdata
17:08:56.253  Add     2 17 pizero.local.                    1   1 192.168.137.192

C:\Users\Noralf>ping -4 pizero.local

Pinging pizero.local [192.168.137.192] with 32 bytes of data:
Reply from 192.168.137.192: bytes=32 time<1ms TTL=64
```

This is a log of what goes on behind the scenes when this service is started (dhcpcd debugging enabled): https://gist.github.com/notro/281e4f0b3781cc855736
During my testing it has usually taken >1 minute to get a dhcp address.

I follwed this guide to install the USB RNDIS driver on Windows 8.1 (not the ip address part): http://www.walchem.com/literature/Controllers/WM/180613_A_USB%20Driver_Windows%208.pdf
I installed Bonjour 3.0 extracted from the iTunes installer: https://learn.adafruit.com/bonjour-zeroconf-networking-for-windows-and-linux/overview

NicoHood, 2016-04-04:
I really like the idea. That would make things a lot simpler, because you do not need a serial port to setup the initial configuration. Any updates on this?

notro, 2016-07-12:
There's a better way to do this using the ```modules-load=``` kernel commandline parameter understood by systemd.
Man page: https://www.freedesktop.org/software/systemd/man/systemd-modules-load.service.html

Ref: http://blog.gbaman.info/?p=791

closed 2016-07-12 by notro