RPi-Distro/repo #21: "Homepage" link for 'raspberrypi-sys-mods' is dead

allanlewis, 2016-02-23:
The package `raspberrypi-sys-mods` has the following description, as shown in `aptitude`:

> Various modifications to improve the performance or user experience.
> Homepage: https://github.com/RPi-Distro/raspberrypi-sys-mods

However, the "homepage" link is dead and the repository doesn't exist on GitHub. If the source no longer exists, perhaps the package should be removed as obsolete? (I'm not sure what Debian/Raspbian/RPi policy is on this.)

_I originally raised this in [Raspbian Launchpad](https://bugs.launchpad.net/raspbian/+bug/1547876), but was [advised](https://bugs.launchpad.net/raspbian/+bug/1547876/comments/1) by ['diederek'](https://launchpad.net/~didi-debian) to raise it here instead._

XECDesign, 2016-02-25:
Thanks, should be there now.

closed 2016-02-25 by XECDesign

allanlewis, 2016-02-25:
Thanks, @XECDesign - it would be nice if that repo at least had a description, if not a read-me...