RPi-Distro/repo #25: Missing github repos

lurch, 2016-03-31:
I'm not sure if this is the best place to report this issue, but I dunno if there's somewhere more suitable.

During a recent `sudo apt-get dist-upgrade` I noticed there was a new `pishutdown` package, so I did `apt-cache show pishutdown` and it says `Homepage: http://github.com/raspberrypi/pishutdown`, but that URL just 404s.

Also, I noticed that `cat /boot/issue.txt` says `Generated using Pi-gen, https://github.com/RPi-Distro/Pi-gen, stage4` and that repo is missing too. (I assume this latter project is an updated fork of https://github.com/RPi-Distro/spindle ?)

XECDesign, 2016-03-31:
pishutdown is there, just private for now. The source itself should be in apt.

Pi-gen hasn't been published yet.

spl237, 2016-04-02:
Source code repos will generally be kept private until the relevant apps have been released. I'm not sure why people would want the source of something they don't even know exists yet...

Patience, as they say, is a virtue. :)

closed 2016-04-02 by spl237

XECDesign, 2016-04-02:
@spl237 pishutdown is in apt already.

spl237, 2016-04-02:
In that case, I'll open up the repo when I'm next in the office...

lurch, 2016-04-02:
@XECDesign Any idea if / when Pi-gen will get published? I can imagine people would find it quite useful for e.g. creating customised Raspbian-Lite images?

ZanderBrown, 2016-04-03:
@lurch it would be so useful particularly if a nice interface was available for it so you could have your SD card already to go as a webserver or whatever before you've even plugged your pi in.

lurch, 2016-04-03:
@ZanderBrown I have no insider info, but I'm expecting it'll be a purely script-based application, with the only "user interface" being a configuration text-file you'll need to edit.

ZanderBrown, 2016-04-04:
My point was more that once it was available a wizard for generating said file could be made