RPi-Distro/repo #28: Default Raspbian Jessie install spams syslog with "rsyslogd-2007: action 'action 17' suspended, next retry is

DanTup, 2016-04-12:
My syslog contains loads of these intermittently:

```
Apr 12 04:30:18 raspberrypi rsyslogd-2007: action 'action 17' suspended, next retry is Tue Apr 12 04:31:48 2016 [try http://www.rsyslog.com/e/2007 ]
Apr 12 04:37:37 raspberrypi rsyslogd-2007: action 'action 17' suspended, next retry is Tue Apr 12 04:39:07 2016 [try http://www.rsyslog.com/e/2007 ]
Apr 12 04:43:15 raspberrypi rsyslogd-2007: action 'action 17' suspended, next retry is Tue Apr 12 04:44:45 2016 [try http://www.rsyslog.com/e/2007 ]
Apr 12 04:48:56 raspberrypi rsyslogd-2007: action 'action 17' suspended, next retry is Tue Apr 12 04:50:26 2016 [try http://www.rsyslog.com/e/2007 ]
Apr 12 04:53:42 raspberrypi rsyslogd-2007: action 'action 17' suspended, next retry is Tue Apr 12 04:55:12 2016 [try http://www.rsyslog.com/e/2007 ]
Apr 12 04:55:54 raspberrypi rsyslogd-2007: action 'action 17' suspended, next retry is Tue Apr 12 04:57:24 2016 [try http://www.rsyslog.com/e/2007 ]
Apr 12 04:58:54 raspberrypi rsyslogd-2007: action 'action 17' suspended, next retry is Tue Apr 12 05:00:24 2016 [try http://www.rsyslog.com/e/2007 ]
Apr 12 05:06:17 raspberrypi rsyslogd-2007: action 'action 17' suspended, next retry is Tue Apr 12 05:07:47 2016 [try http://www.rsyslog.com/e/2007 ]
Apr 12 05:11:04 raspberrypi rsyslogd-2007: action 'action 17' suspended, next retry is Tue Apr 12 05:12:34 2016 [try http://www.rsyslog.com/e/2007 ]
Apr 12 05:17:01 raspberrypi rsyslogd-2007: action 'action 17' suspended, next retry is Tue Apr 12 05:18:31 2016 [try http://www.rsyslog.com/e/2007 ]
Apr 12 05:20:06 raspberrypi rsyslogd-2007: action 'action 17' suspended, next retry is Tue Apr 12 05:21:36 2016 [try http://www.rsyslog.com/e/2007 ]
Apr 12 05:25:54 raspberrypi rsyslogd-2007: action 'action 17' suspended, next retry is Tue Apr 12 05:27:24 2016 [try http://www.rsyslog.com/e/2007 ]
Apr 12 05:28:48 raspberrypi rsyslogd-2007: action 'action 17' suspended, next retry is Tue Apr 12 05:30:18 2016 [try http://www.rsyslog.com/e/2007 ]
Apr 12 05:36:44 raspberrypi rsyslogd-2007: action 'action 17' suspended, next retry is Tue Apr 12 05:38:14 2016 [try http://www.rsyslog.com/e/2007 ]
Apr 12 05:41:28 raspberrypi rsyslogd-2007: action 'action 17' suspended, next retry is Tue Apr 12 05:42:58 2016 [try http://www.rsyslog.com/e/2007 ]
Apr 12 05:48:51 raspberrypi rsyslogd-2007: action 'action 17' suspended, next retry is Tue Apr 12 05:50:21 2016 [try http://www.rsyslog.com/e/2007 ]
Apr 12 05:55:54 raspberrypi rsyslogd-2007: action 'action 17' suspended, next retry is Tue Apr 12 05:57:24 2016 [try http://www.rsyslog.com/e/2007 ]
Apr 12 06:05:18 raspberrypi rsyslogd-2007: action 'action 17' suspended, next retry is Tue Apr 12 06:06:48 2016 [try http://www.rsyslog.com/e/2007 ]
Apr 12 06:11:17 raspberrypi rsyslogd-2007: action 'action 17' suspended, next retry is Tue Apr 12 06:12:47 2016 [try http://www.rsyslog.com/e/2007 ]
Apr 12 06:17:01 raspberrypi rsyslogd-2007: action 'action 17' suspended, next retry is Tue Apr 12 06:18:31 2016 [try http://www.rsyslog.com/e/2007 ]
Apr 12 06:25:01 raspberrypi rsyslogd-2007: action 'action 17' suspended, next retry is Tue Apr 12 06:26:31 2016 [try http://www.rsyslog.com/e/2007 ]
Apr 12 06:34:05 raspberrypi rsyslogd-2007: action 'action 17' suspended, next retry is Tue Apr 12 06:35:35 2016 [try http://www.rsyslog.com/e/2007 ]
Apr 12 06:43:34 raspberrypi rsyslogd-2007: action 'action 17' suspended, next retry is Tue Apr 12 06:45:04 2016 [try http://www.rsyslog.com/e/2007 ]
Apr 12 06:48:26 raspberrypi rsyslogd-2007: action 'action 17' suspended, next retry is Tue Apr 12 06:49:56 2016 [try http://www.rsyslog.com/e/2007 ]
Apr 12 06:50:05 raspberrypi rsyslogd-2007: action 'action 17' suspended, next retry is Tue Apr 12 06:51:35 2016 [try http://www.rsyslog.com/e/2007 ]
```

Seems to affect many, has been mentioned in the forums:
https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=134971

It was reported in Debian:

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=742113

but they said it was an rsyslog issue, and reported here:

https://github.com/rsyslog/rsyslog/issues/35

and there are some comments on there suggesting it's just the default config writing to xconsole:

> It's a configuration issue with the distro then, nothing wrong with rsyslog. Commented out the offending lines with much success. Thanks!

I don't really know which of these is correct, but I thought it'd be worth posting here in case it's something you can fix. I'm trying to reduce the amount of crap being written to my logs so that a) I can keep an eye on them (I'm a Linux noob, so trying to understand what's going on/wrong) and b) seems like lots of needless writing on an SD card could reduce its life.

XECDesign, 2016-04-21:
Does it go away if you install x11-apps?

DanTup, 2016-04-21:
@XECDesign I appear to already have it:

```
danny@raspberrypi:~$ sudo apt install x11-apps
Reading package lists... Done
Building dependency tree        
Reading state information... Done
x11-apps is already the newest version.
```

XECDesign, 2016-04-22:
We haven't changed the default rsyslog config in any way, so it shouldn't differ from from Debian. 

Looks like you've looked into it much more than I have. If you can figure out what that message is and how to make it go away without negatively affecting functionality, it would certainly help.

DanTup, 2016-04-22:
I'm a bit of a noob, I'm afraid! The other posts suggest it's caused by this:

```
daemon.*;mail.*;\
    news.err;\
    *.=debug;*.=info;\
    *.=notice;*.=warn   |/dev/xconsole
```

specifically, the xconsole at the end. Commenting these lines out apparently fixes it. I haven't tried it yet because I'm not sure I understand the implications, or why those lines are there if xconsole is not! :(

XECDesign, 2016-04-22:
At least on your install, xconsole is installed, and on all installs /dev/xconsole is present, so I am not sure what that would have to do with it.

Whatever the issue is, I don't think it's our fault. We're using stock rsyslog from Raspbian without modifications, which should be the same as it is in Debian.

DanTup, 2016-04-22:
Ok, then I guess this case can be closed. I'll see if maybe @rgerhards can help over in https://github.com/rsyslog/rsyslog/issues/35

Thanks for responses!

closed 2016-04-22 by DanTup

bcutter, 2016-05-03:
Hi,

I was lucky to find something here (also Pi 2 on Raspbian = Debian Jessie, flooded with ton´s of those "rsyslogd-2007: action 'action 17' suspended, next retry is Tue May  3 12:08:31 2016 [try http://www.rsyslog.com/e/2007 ]"messages). I tried to follow the discussion here, but am not sure what the final result is.

Is it "you´re wrong here" aka "not my department"? If so, maybe @DanTup can instruct me on his further investigations on this (outside this topic).

DanTup, 2016-05-03:
I removed the bottom part of `rsyslog.conf` to stop it with this command:

    sed -i '/# The named pipe \/dev\/xconsole/,$d' /etc/rsyslog.conf

This deletes everything from "# The named pipe /dev/xconsole" to the end of the file.

Hope that helps!

bcutter, 2016-05-05:
Well, I monitored syslog for a while using "sudo watch -n1 "tail /var/log/syslog"", the messages seem to have gone. Hope there´ll be no side effects. Thanks so far!

andrum99, 2017-04-07:
I've hit this bug as well. See the bottom part of this bug report for some background:

https://github.com/rsyslog/rsyslog/issues/35

In particular the comment from dadidelang: "the era of spitting logs out to a console and having someone see them is long over. ..."

And a link to the relevant Debian commit that fixes it:

https://anonscm.debian.org/cgit/collab-maint/rsyslog.git/commit/?id=67bc8e5326b0d3564c7e2153dede25f9690e6839

In other words - it is a bug in the Debian configuration of rsyslogd which is fixed in Stretch.

I would therefore like to add my +1 to fixing this in Raspbian Jessie. Please reopen this bug.

rsiesta, 2017-05-21:
May 21 11:17:01 davis rsyslogd-2007: action 'action 17' suspended, next retry is Sun May 21 11:17:31 2017 [try http://www.rsyslog.com/e/2007 ]
May 21 12:13:40 davis rsyslogd0: action 'action 17' resumed (module 'builtin:ompipe') [try http://www.rsyslog.com/e/0 ]
May 21 12:13:40 davis rsyslogd-2359: action 'action 17' resumed (module 'builtin:ompipe') [try http://www.rsyslog.com/e/2359 ]

If you look at the /etc/rsyslog.conf file at the end you will see the piping to /dev/xconsole, note the comments and the fact the piping to the console is commented out, probably should just comment out the ones to xconsole, or fix whatever is breaking the pipe. When I cat'd /dev/xconsole I saw there were entries, after the cat the above message appeared in syslog and the others went away. For whatever reason touching it fixed the broken pipe.

rsiesta, 2017-05-21:
I have never had to look at the xconsole output so I just commented out the command, for those that want to take this approach   comment the following out of /etc/rsyslog.conf

daemon.*;mail.*;\
        news.err;\
        *.=debug;*.=info;\
        *.=notice;*.=warn       |/dev/xconsole