RPi-Distro/repo #29: Badlock

Skywalker-11, 2016-04-15:
A security vulnerability was found in Samba which is called "Badlock". There already exist security patches for debian since the 12th april (2:4.2.10+dfsg-0+deb8u2).

In the rasbian repository there still only is version "2:4.1.17+dfsg-2+deb8u2" available.

This is a security vulnerability that also has an CVE: CVE-2016-2118

So it would be really nice if you can upgrade the package as fast as possible.

More information can be found at http://badlock.org/

plugwash, 2016-04-21:
It looks like when Debian updated the samba/ldb/tevent packages they didn't set proper build-depends in the samba and ldb packages, this resulted in our autobuilders trying to build them all at the same time and the ldb and samba packages failing to build.

I've just rescheduled ldb, once that builds successfully I will reschhedule samba.

plugwash, 2016-04-21:
Fixed version should be in the repo now.

Skywalker-11, 2016-04-22:
Thanks. New version is online.

closed 2016-04-22 by Skywalker-11