RPi-Distro/repo #30: fixed in netplug 1.2.9.2-2

adrianrieck, 2016-04-19:
With the current repo version of netplug  1.2.9.2-1 there is a bug where it dies because of an 'unexpected state DOWNANDOUT for UP'

When it happens this shows up in my logs:
daemon.log.1:Apr 13 23:14:42 raspberrypi netplugd[944]: eth0: unexpected state DOWNANDOUT for UP

The bug is described here as well:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=652418

It does appear that it has been fixed since netplug version 1.2.9.2-2.

Please consider updating the Raspbian repo with the fixed version. 

Thank you!

XECDesign, 2016-05-01:
A little reluctant to start trying to fix bugs in standard debian packages. Unless we've introduced it or it's problem that affects a significant portion of users, we'll generally not touch it.

adrianrieck, 2016-05-02:
I suppose you want to keep things on par with Debian. I guess we will have to build it ourselves until at least Debian Stretch is released?  Thank you for your time.

diederikdehaas, 2016-05-02:
Or use the Debian bug report to request a backport of the fix to stable.
There's no guarantee that it will be done, but you can try it.

XECDesign, 2016-05-02:
It's not even just that. It's also a bit of a rabbit hole. There are many Debian packages with many known problems that have been fixed in stretch or sid. We can't go backporting everything for a variety of reasons. It is more likely than not to introduce new problems, new dependencies, new conflicts, it will take resources away from our own packages, then it's up to us to keep track of security fixes for those packages and so on.

diederikdehaas, 2016-05-02:
I meant requesting the Debian maintainer to fix it in Debian Stable. 
**If** that happens, it should trickle down automatically to Raspbian(.org) Jessie.

XECDesign, 2016-05-02:
Sorry, that was a response to adrianrieck that it's not just that we're trying to keep things on par with Debian.

closed 2016-05-03 by XECDesign