RPi-Distro/repo #31: Raspberry Pi 3 serial ports (still) not working !

PratikSK, 2016-05-20:
Hello there,

I just purchased RPi3 a week before and trying to connect GSM SIM 800 module using serial ports (pin 8, 10).
Initially I am trying check serial ports by connecting pin 8-10 and using minicom.
I have tried several tricks from no of threads but unfortunately none of them seems to work for me.

I have updated device using :
sudo apt-get update
sudo apt-get upgrade
sudo apt-get dist-upgrade
sudo rpi-update

Changes I have made : 
1. Disable Bluetooth ports and bring serial back to AMA0 (instead of S0) using thread [https://openenergymonitor.org/emon/node/12311](url) 
2. Then I added following famous lines to my /boot/config.txt
   core_freq=250
   dtoverlay=pi3-disable-bt
   force_turbo=1
   console=serial1

sudo minicom -D dev/ttyAMA0 -b38400 
After this it is showing some garbage value. (precisely it's showing some raspberrypi login with order of letters broken)
And I cannot type if I remove wire connecting pins 8-10. Connecting 8-10 again, I can type again.

I have tried several different baud rates etc but ..

Can anyone pleeeaase conclude this topic! Thanks a lot in advance.
Please let me know what else I need to do.

pelwell, 2016-05-20:
To get a console on /dev/ttyS0, put `enable_uart=1` in config.txt.

To make /dev/ttyAMA0 available for application use, stop Bluetooth using it by running `sudo systemctl disable hciuart`.

You can then remove all of your four setting from config.txt, so no "core_freq=250", "dtoverlay=pi3-disable-bt", "force_turbo=1" or "console=serial1".

PratikSK, 2016-05-20:
Thanks for instant reply.

I tried above. 
After adding enable_uart=1
I will run this using:  sudo minicom -D /dev/ttyS0 -b9600
right.? I tried this but not getting output.

Meanwhile, I tried to change frequency core_frequencies in range 400,450.
It showed output for 450 a couple of times. (i.e. It was repeating the letters I was typing)
But it wasn't consistent. I was getting some garbage value sometimes or not working at all with same setting at other times.

Thanks,

pelwell, 2016-05-20:
You are confusing console use and application use - you don't want the console and minicom to both try to use the UART at the same time.

My instructions were to disable Bluetooth to leave ttyAMA0 available for minicom, and to enable ttyS0 for console I/O.

PratikSK, 2016-05-21:
I tried according to that, still not getting the results.
Can you tell / direct me to a step by step procedure where I will start from scratch and get the things done. 
Thanks in advance.

pelwell, 2016-05-21:
Tell me what you have done and what results you get.

pelwell, 2016-05-21:
Don't bother - it's obvious what's wrong. As I said before, you can't use the same serial port for two things at the same time. What I didn't say is that there is only one set of pins on the header that the serial ports can be mapped to. Either ttyS0 or ttyAMA0 can be accessed, but not both (unless you have a Compute Module).

PratikSK, 2016-05-21:
I install jessie released on 18th March 2016. 
Then updated it.
Installed minicom.

Then I put following in config file (by referring various forums)
core_freq=250
dtoverlay=pi3-disable-bt
force_turbo=1
console=serial1

Connected pin 8 and 10 to each other,

Then I used " sudo minicom -D dev/ttyAMA0 -b9600 " for minicom. Tried to change baud rates etc.
But nothing worked. (It was showing garbage some times, nothing at other time)

Then I randomly increased core_freq, 
I got it working on core_freq 450. I was getting back whatever I was typing. I was still getting garbage letters also.
This output was not steady. With same settings, it didn't worked many times after that.  

Then I tried your way, I removed things I had added to config and put enable_uart=1 over there.
Rebooted.
Then used this " sudo systemctl disable hciuart "
I then tried  " sudo minicom -D dev/ttyAMA0 -b9600 "  as well as " sudo minicom -D dev/ttyS0 -b9600 " with different baud rates. 
I am not getting anything back when I type there.

Thanks :)

pelwell, 2016-05-21:
1) Use `dtoverlay=pi3-disable-bt` to free up ttyAMA0 for your own use. If you might want to use Bluetooth you could use `dtoverlay=pi3-miniuart-bt` instead, but you would also need to reduce the baudrate in /lib/systemd/system/hciuart.service to 460800.

2) ttyAMA0 is enabled by default, but there is no harm in including `enable_uart=1`.

3) You don't want anything else to use ttyAMA0 (or ttyS0) so remove all references to console=serial0 and console=serial1 from `cmdline.txt` (where it belongs) and `config.txt` (where it doesn't belong).

4) ttyAMA0 does not change its baudrate with the core clock speed, so you don't need core_freq=250 or force_turbo=1.

5) Now connect pins 8 and 10, boot your Pi and run `sudo minicom -D /dev/ttyAMA0 -b 9600`. You should see your characters looped back.

pelwell, 2016-05-21:
If for some reason that doesn't work, run `raspi-gpio get 14` and `raspi-gpio get 15` to confirm that function a0 is enabled on those pins.

PratikSK, 2016-05-21:
Ok just trying out this one. 
Thanks

PratikSK, 2016-05-21:
I tried above. But not getting characters back.

PratikSK, 2016-05-21:
After running 
raspi-gpio get 14    is  " GPIO 14: level=1 fsel=0 func=INPUT "
I get exactly same output after " raspi-gpio get 15 " as well

func for both of them is INPUT. Is it correct .?

pelwell, 2016-05-21:
No - that isn't correct. You can fix it for this boot by running:
```
raspi-gpio set 14 a0
raspi-gpio set 15 a0
```
But that doesn't explain why it was wrong in the first place. Run `sudo rpi-update` to get the latest version of the firmware and try again.

PratikSK, 2016-05-21:
Ok this is cool. After setting it func has changed to TXD0 and RXD0 but it's still not helping to get characters back. I will rpi-update and report soon
Thanks

PratikSK, 2016-05-21:
My firmware is already up to date.
I had updated it few hours ago!

pelwell, 2016-05-21:
Please post the contents of config.txt and cmdline.txt.

PratikSK, 2016-05-21:
cmdline.txt :

dwc_otg.lpm_enable=0 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait

config.txt:

# For more options and information see
# http://www.raspberrypi.org/documentation/configuration/config-txt.md
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
#dtparam=audio=on
#dtoverlay=pi3-disable-bt
#core_freq=450
#force_turbo=1
#console=serial1
enable_uart=0

pelwell, 2016-05-21:
> enable_uart=0

What is that doing there? I asked you to use `enable_uart=1`. That line has disabled the UART - no wonder it doesn't work. Change that to `enable_uart=1`, reboot and all should be well.

PratikSK, 2016-05-21:
Ow, sorry for that. It was embarrassing!  
I did it and rebooted. Still not getting characters back :(

pelwell, 2016-05-21:
You've also lost the `dtoverlay=pi3-disable-bt`, which probably accounts for the rest of the problems.

Is this a NOOBS installation? Have you been editing the config.txt and cmdline.txt in the wrong partition?

PratikSK, 2016-05-21:
It is not NOOBS installation. I installed using windiskimager by writing image.

I have used dtoverlay=pi3-disable-bt in command line before running minicom.

I think once I will start from scratch from installing the image itself.

pelwell, 2016-05-21:
No - don't do that.

Put `dtoverlay=pi3-disable-bt` in config.txt, as I told you to, and reboot.

PratikSK, 2016-05-21:
Added that line and rebooted ,
not getting char back ..

PratikSK, 2016-06-24:
This problem is solved with above mentioned procedure I.e. by editing config.txt and cmdline.txt files as mentioned above.

Thanks a lot pelwell.

closed 2016-06-24 by PratikSK

gatorreina, 2018-01-02:
After 24 hours of frustration. I am unable to a serial port working with my Element Raspberry Pi 3 running Raspian Jessie. I have added the following to /boot/config.txt as instructed in this text ( I do not need bluetooth). I have connected pins 8 and 10 to their opposite on the serial port. (RX to TX, TX to RX) I have connected the ground and connected the UCC to 3.3n, 5v and nothing -- currently it is still unplugged. 

enable_uart=1
dtoverlay=pi3-disable-bt

$ cat /boot/cmdline.txt
dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p7 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait splash plymouth.ignore-serial-consoles

pelwell, 2018-01-02:
Make sure you are using the correct baud rate and parity setting - UARTs have to agree, otherwise data is dropped or corrupted. It might also be worth trying with the RX/TX lines swapped at one end, just in case the labelling is misleading.

gatorreina, 2018-01-02:
Pellwell, Thanks for the reply. How do I check baud rate and parity settings?

pelwell, 2018-01-02:
`stty -F /dev/ttyAMA0 -a` should do the trick, but I think you need to say a bit more about the device you have connected and how you are trying to communicate with it.

gatorreina, 2018-01-02:
Holy Cow!! It was the pins! This is incredible. I now have pin 8 connected to the pin labelled TX on the serial port and pin 10 connected to the pin labelled RX on the serial port and it now works. Can't believe it!

pelwell, 2018-01-02:
Somebody was being "helpful". Glad you got it sorted.

gatorreina, 2018-01-02:
Thank you very much for the suggestion Pelwell. Would have never expected that.

swaggable, 2018-01-22:
[image]

how to add AT commands, pls help

swaggable, 2018-01-22:
[image]

how to fix this? i want to see my serial if it is working

widapraponco, 2018-02-02:
[image]

hi i've been trying to attach MaxSonar range finder sensor on rpi3 and already follow all the instruction above, but i get this weird character. what i expect is R0764 like this tutorial link https://www.maxbotix.com/Raspberry-Pi-With-Ultrasonic-Sensors074/
can anyone give me an advice?

XECDesign, 2018-02-02:
General queries like that are usually handled on the forum.
https://www.raspberrypi.org/forums/

giftbalogun, 2023-12-21:
Hi Everyone, I Hope I am not late. 
I am trying to get the WaveShare SIM7600G-H 4G to work with the Pi3 Model B, I have followed waveshare docs I cannot get it to work on the PI.
When I even try the Example Command,  for SMS, O just get this

pi@raspberrypi:~/SIM7600X-4G-HAT-Demo/Raspberry/python/SMS $ sudo python3 SMS.py
SIM7600X is starting:
SIM7600X is ready
Sending Short Message Test:
Setting SMS mode...

Please someone help me out,
Thanks in Advance

popcornmix, 2023-12-21:
Why are you posting on an unrelated, closed, 5 year old issue?
You might have more luck on the [forums](https://forums.raspberrypi.com/index.php)