RPi-Distro/repo #34: raspiviyuv isn't symlinked into $PATH

lurch, 2016-06-21:
I noticed recently that `raspividyuv` is available in the https://github.com/raspberrypi/userland and https://github.com/raspberrypi/firmware repos, but it isn't available in the $PATH by default. So I did a bit of investigation...

Looking at the output of `dpkg -L libraspberrypi-bin` I see:
```
/opt/vc/bin/raspistill
/opt/vc/bin/raspiyuv
/opt/vc/bin/raspivid
/opt/vc/bin/raspividyuv
...
/usr/bin/raspistill
/usr/bin/raspiyuv
/usr/bin/raspivid
```
with the files in `/usr/bin/` simply being symlinks to the files in `/opt/vc/bin/`.
I guess the `libraspberrypi-bin` package also ought to add a symlink from `/usr/bin/raspividyuv` to `/opt/vc/bin/raspividyuv`?

XECDesign, 2016-06-21:
Yes, that makes sense. Will do.

XECDesign, 2016-06-22:
Fixed by https://github.com/RPi-Distro/firmware/commit/266a691af77ef9f4f9d8bdffac5fa0a7b7153f4f

closed 2016-06-22 by XECDesign

lurch, 2016-06-22:
Great! :)