RPi-Distro/repo #35: Kodi out of date and broken in testing

PanderMusubi, 2016-06-24:
Not sure where to report this. Was reported here https://bugs.launchpad.net/raspbian/+bug/1571196 but is not followed up. Since kodi is in the rasberry repo for Raspbian, not the raspbian repo for Raspbian, I thought perhaps it belows here

PanderMusubi, 2016-06-29:
See launchpad for workaround.

XECDesign, 2016-07-24:
We don't have 'testing' in raspberrypi.org. However, we should update our kodi packages soon.

closed 2016-07-24 by XECDesign

PanderMusubi, 2016-07-25:
this one is testing:

    deb [arch=armhf] http://mirrordirector.raspbian.org/raspbian/ testing main contrib non-free rpi

diederikdehaas, 2016-07-25:
raspbian.org != raspberrypi.org

PanderMusubi, 2016-07-25:
Thanks. So, bug reports for that go only to https://launchpad.net/raspbian ?

diederikdehaas, 2016-07-25:
Bug reports for packages coming from raspbian.org, yes.
The statement you made in your first post in this issue is (completely) correct.