RPi-Distro/repo #37: Submitting Packages

munawarb, 2016-07-12:
I have seen on some forums that the way to submit packages to be included in the RPi Repo so that users can access using apt-get is to submit a bug called an "intent to package." Do I submit the bug report here or elsewhere? Thanks for your help!

XECDesign, 2016-07-24:
There's no official procedure to submit packages to the raspberrypi.org repo. If it's not a pi-specific package, it needs to be submitted to Debian and will eventually make it into raspbian.org. If it's a particularly useful package, then we can take it earlier into our repo.

If it's a pi-specific package that wouldn't be accepted by Debian, then you can get in touch (here is okay) and provide more information.

XECDesign, 2016-08-10:
Closing for inactivity.

closed 2016-08-10 by XECDesign