RPi-Distro/repo #39: Wrong device initialized by pi-bluetooth

rgov, 2016-08-18:
I have a RPi 3 Model B running Raspbian Jessie, although it is somewhat of a Frankenstein system.

I installed `pi-bluetooth` but was struggling to get the controller to work. Then I found [this comment](https://github.com/raspberrypi/linux/issues/1314#issuecomment-190785413) proposing a `hciattach` command that worked for me.

It looks like the `pi-bluetooth` is executing

```
/usr/bin/hciattach /dev/serial1 bcm43xx 921600 noflow -
```

but there is no `/dev/serial1` on my device. Replacing `serial1` with `ttyAMA0` in that file addressed the issue.

XECDesign, 2016-08-18:
Seems like raspberrypi-sys-mods is missing as a dependency. Do you have it installed?

rgov, 2016-08-18:
I didn't have it installed. I installed it, and now I have `/dev/serial1` and it seemed to work on the next boot. Thanks!

XECDesign, 2017-01-23:
Should be fixed here https://github.com/RPi-Distro/pi-bluetooth/commit/b9d9145de0564380f959d6c48587c189367258ec

closed 2017-01-23 by XECDesign