RPi-Distro/repo #40: raspberrypi-bootloader should not depend on raspberrypi-kernel

ali1234, 2016-08-24:
It is perfectly reasonable to use a custom kernel without having raspberrypi-kernel installed. But the dependency does not allow this.

XECDesign, 2016-08-24:
I worry that removing the dependency may leave some people without any kernel at all. Or worse, it may consider it an orphan package. Since it was all a single package in the past, the dependency is there to ensure that when apt/dpkg upgrades from the old version, the kernel package is installed.

Will let you know if it's possible after I check exactly what apt would do.

ali1234, 2016-08-24:
Perhaps making it a "recommends" would solve that problem, while still allowing power user to install one without the other.

ali1234, 2016-08-24:
In case you are wondering, this is why I need it:

https://github.com/ali1234/debirf-raspi/

The modules symlinked in modules/ are run in order with run-parts. Problem is that k0_bootloader pulls in the kernel too. This isn't a problem if you have symlinked k0_rpi-kernel, it will just see the kernel is already installed and do nothing. But if you want a custom kernel it's a problem. I'd have to manually delete the unwanted kernel files, or mess with dpkg --force-all, or equivs.

XECDesign, 2016-10-07:
Just flashed an old image and added modified firmware kernel packages to a local repo. `apt-get dist-upgrade` seems to do the right thing and installs raspberrypi-kernel. So I don't think many people will accidentally end up without a kernel as a result of this change. Still somewhat worried that there might be some people doing something unusual which results in the package being removed or not installed.

XECDesign, 2016-10-17:
Having played around with it a bit, I have failed to find a situation where it causes problems, unless `--no-install-recommends` is used when initially installing the package. I knew that adding recommends packages does not automatically install them when the user dist-upgrades, so I was worried that since it's an 'auto' marked package, apt might decide it's an orphaned package and remove it, even if moved to recommends.
Should be fixed with the next package update.
https://github.com/RPi-Distro/firmware/commit/ea475c1dcc3757c51ba7cc6fa54d105ae2842347

XECDesign, 2016-10-24:
Fixed in the latest release

closed 2016-10-24 by XECDesign