RPi-Distro/repo #43: Ugly colour banding with gstreamer video

herrernst, 2016-10-04:
I've tried to use `gst123` or `gst-launch playbin` (which is then probably using glimagesink) to play videos on the terminal instead of omxplayer (with gstreamer1.0-omx hardware acceleration). I works almost good (no stuttering), but I get considerable colour banding, which does not occur with omxplayer. Is there anything I can do about it? Thanks

XECDesign, 2016-10-04:
Unfortunately gstreamer is no longer maintained.

gkreidl, 2016-12-14:
That's not true and a lot of Raspbian packages depend on gstreamer packages.

XECDesign, 2016-12-14:
I should've worded that clearer. The collabora omx patches are not maintained.

gkreidl, 2016-12-14:
I hope you will leave it unchanged, at least, because the current version contains some patches which I have sent them to fix some bad memory leaks.

XECDesign, 2016-12-14:
No plans to touch it in jessie. No idea what will happen in stretch.

closed 2017-01-23 by XECDesign

herrernst, 2017-01-23:
I just wanted to say that I am not happy that gstreamer omx will not be maintained anymore, gstreamer being a widely used library.

XECDesign, 2017-01-23:
I know, it's not the best situation. But it looks like gst-omx is still maintained upstream, so it may be worthwhile looking at how stable that is and whether it can be merged back into Raspbian.

gkreidl, 2017-01-23:
And I definitely need it. It's the backbone of my real time TV transcoding package.

XECDesign, 2017-01-23:
@gkreidl, how does upstream compare to what's in the repo as far as stability and performance goes?

gkreidl, 2017-01-23:
I don't know. I'm still using Jessie version. There was a memory leak in gst-omx and I've sent a fix from upstream to the collabora people and they applied that patch (it's in the current version). I also tried to compile a 1.2 version which never worked. There's a script on github to compile latest versions and someone on the forum said he got it to work including gst-omx, but I haven't tried it myself.
https://gist.github.com/sphaero/02717b0b35501ad94863

fnoop, 2017-02-05:
I'm trying to get upstream (1.10.3) gst-omx working.  I get it compiled fine but it hangs when doing any encoding.  Are the collabora patches available somewhere to look at?

XECDesign, 2017-02-05:
@fnoop http://archive.raspberrypi.org/debian/pool/main/g/gst-omx1.0/gst-omx1.0_1.0.0.1-0+rpi18.debian.tar.xz

fnoop, 2017-02-05:
Thanks :)  I'm surprised upstream doesn't work with raspberry, there's so much reference in recent changelogs.

gkreidl, 2017-02-05:
Obviously a lot of upstream code (gst_omx) has never been tested on a Raspberry Pi. I couldn't even get version 1.2 (including patches from collabora) to work.

fnoop, 2017-02-06:
What source does this apply to?  I've tried applying to the original 1.0.0 source release and half the patch chunks fail.

fnoop, 2017-02-06:
Ah found it the archive directory, sorry:
http://archive.raspberrypi.org/debian/pool/main/g/gst-omx1.0/gst-omx1.0_1.0.0.1.orig.tar.xz

gkreidl, 2017-02-06:
I pulled the sources from the collabora git.

fnoop, 2017-02-06:
OK found branch rpi-1.0.0.1 in https://git.collabora.com/cgit/raspberry-pi/gst-omx.git/ that seems to be patched source.  Sigh, this doesn't compile against recent gstreamer releases (I'm trying 1.10.3).  
Really sad that raspberry is completely stalled on gstreamer..