RPi-Distro/repo #46: debhelper templates for maintainer scripts of packages containing kernel modules

mforkel, 2016-10-26:
While packaging some kernel modules, I noticed that the debhelper templates `/usr/share/debhelper/autoscripts/postinst-modules` and `/usr/share/debhelper/autoscripts/postrm-modules` don't seem to be appropriate for Raspbian. Both templates refer to `/boot/System.map-#KVERS#`, but System.map files do not exist in Raspbian. As a result, the calls to depmod created by `dh_installmodules` and inserted in to the maintainer scripts by `dh_installdeb` will not be executed after installation or removal of the kernel modules.

XECDesign, 2016-10-27:
Are these DKMS modules or do they come pre-compiled? I seem to recall DKMS working, but if there are pre-compiled, you might have trouble keeping up with raspberrypi-kernel.

mforkel, 2016-10-27:
Currently, I'm not using DKMS. I recompile the modules whenenver there is a new version of package `raspberrypi-kernel` and create corresponding versions of my packages. Each of them `depends` on that specific version of `raspberrypi-kernel`. In order to have `depmod` executed during an upgrade, I replace the tests for `/boot/System.map-#KVERS#` by tests for `/lib/modules-#KVERS#` in the maintainer scripts.

XECDesign, 2016-10-27:
I think adding an override for `dh_installmodules` with your own version might be the way to go. Perhaps it could call `dh_installmodules` as normal and then sed replace the affected strings?

We grab the kernel and firmware as is from raspberrypi/firmware. Our kernel package is too different to Debian's, so I wouldn't expect these things to work.

If @popcornmix would like to add a /boot/System.map-$KVERS, then that could be a fix. Another option is to make the package add the file in postinst.

mforkel, 2016-10-27:
My fix is very similar to what you suggested. `dh_installmodules` just adds text snippets to `debian/*.postinst.debhelper` (and friends) and `dh_installdeb` inserts those into the maintainer scripts. So I do
```
override_dh_installdeb:
	dh_installdeb
	rpi-fix-maintainer-scripts
```
And `rpi-fix-maintainer-scripts` uses `sed` to replace the references to `System.map-*`.

I think a more general solution might be a patch for the Raspbian version of package `debhelper` that replaces the references to `System.map-#KVERS#`. in `/usr/share/debhelper/autoscripts/postinst-modules` and `/usr/share/debhelper/autoscripts/postrm-modules`.

XECDesign, 2016-10-27:
That's not the best approach because raspberrypi-kernel isn't the only kernel you can run. Raspbian also provides a kernel and I wouldn't want to break their changes. I think this is a wontfix until there is strong motivation to get our kernel in line with Debian's.

mforkel, 2016-10-27:
That's true.

closed 2017-01-23 by XECDesign