RPi-Distro/repo #47: usb_modeswitch udev-rules doesn't work in Rasbian Lite

mcgurk, 2016-11-10:
(Raspbian Jessie Lite, Version:September 2016, Release date:2016-09-23, Kernel version:4.4)

Huawei 3G modem doesn't automatically trigger succesfully usb_modeswitch in Raspbian Lite.

I tracked down problem to udev-rule parameters.
%b is interpreted differently in /lib/udev/rules.d/40-usb_modeswitch.rules.
Line:
ATTRS{idVendor}=="12d1", ATTR{bInterfaceNumber}=="00", ATTR{bInterfaceClass}=="08", RUN+="usb_modeswitch '%b/%k'"
/var/log/syslog (after sudo udevadm control --log-priority=debug):
Lite: Oct 19 17:10:40 raspberrypi systemd-udevd[705]: starting 'usb_modeswitch '1-1.4/1-1.4:1.0''
Desktop: Oct 19 19:22:26 raspberrypi systemd-udevd[910]: starting 'usb_modeswitch '/1-1.4:1.0''

If I start same command by hand with same parameter as in desktop, it works:
sudo /lib/udev/usb_modeswitch /1-1.4:1.0

I can solve problem by changing '%b/%k' to '%k' in /lib/udev/rules.d/40-usb_modeswitch.rule, but I'm not sure if that it is right solution.

XECDesign, 2016-11-10:
Is this the behaviour the same as reported in #12 or are they separate issues?

mcgurk, 2016-11-10:
I think that this is separate issue. This issue comes before any network connection attempt or dhcp stuff.

leolobato, 2016-11-10:
This issue means that when a Huawei 3G modem (which is fairly common) is plugged in, it is not recognized as a modem.

I can confirm that changing `%b/%k` to `%k` on /lib/udev/rules.d/40-usb_modeswitch.rule for Huawei fixes the issue. After a reboot, the modem is recognized.

ralienpp, 2017-02-15:
I encountered the same situation and the changing `%b/%k` to `%k` did the trick; I found that out after trying many other things.

This is very unlikely to be related to `dhcpd` because I disabled `dhcpd` on the system where I dealt with this problem.

XECDesign, 2017-09-08:
Seems to work without problem on stretch lite and desktop.

closed 2017-09-08 by XECDesign