RPi-Distro/repo #49: xserver-xorg-legacy Not Installable

bnielsen1965, 2016-12-22:
Cannot install the xserver-xorg-legacy package on an updated Raspbian system...

> The following packages have unmet dependencies:
>  xserver-xorg-legacy : Breaks: x11-common (< 1:7.7+10~) but 1:7.7+7 is installed.

Cannot upgrade x11-common using apt-get to a version that will allow installation of xserver-xorg-legacy.

XECDesign, 2017-01-12:
Should be resolved

closed 2017-01-12 by XECDesign

bnielsen1965, 2017-01-12:
Confirmed, fixed. The package now installs as expected and my package that depends on legacy support now works as expected. A big thanks!

XECDesign, 2017-01-12:
Thanks for confirming.

ybendou, 2018-07-13:
Seems that the problem is up again, cannot install xserver-xorg-legacy package on amd64 :+1: 
The following packages have unmet dependencies:
 xserver-xorg-legacy : Breaks: x11-common (< 1:7.7+10~) but 1:7.7+7 is to be installed.

I can't upgrade too to a newer version of x11-common with apt-get