RPi-Distro/repo #50: Cannot upgrade `raspberrypi-sys-mods` to 20161220

nisargjhaveri, 2017-01-05:
```
Preparing to unpack .../raspberrypi-sys-mods_20161220_all.deb ...
Error while loading /var/lib/dpkg/tmp.ci/preinst: Exec format error
dpkg: error processing archive /var/cache/apt/archives/raspberrypi-sys-mods_20161220_all.deb (--unpack):
 subprocess new pre-installation script returned error exit status 1
Error while loading /var/lib/dpkg/info/raspberrypi-sys-mods.postinst: Exec format error
dpkg: error while cleaning up:
 subprocess installed post-installation script returned error exit status 1
Errors were encountered while processing:
 /var/cache/apt/archives/raspberrypi-sys-mods_20161220_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

XECDesign, 2017-01-05:
This looks like an issue with a corrupted sd card rather than with raspberrypi-sys-mods.

nisargjhaveri, 2017-01-05:
Oh. Okay. That could've never occurred to me.

Side question: How do I fix it? If it is fixable without re-flashing SD card or getting a new one.

XECDesign, 2017-01-05:
Can't help you with that one, but you might want to ask on the forum. I suspect the answer people will give is to start again.

closed 2017-01-05 by XECDesign