RPi-Distro/repo #51: firmware-brcm80211 breaks wlan0 on Raspbian

fkbreitl, 2017-01-06:
I would like to report issue https://github.com/raspberrypi/linux/issues/1780:

Package: firmware-brcm80211
Version: 20161130-2 (testing)

After upgrading Raspbian from jessie to testing wlan0 disappears on a Raspberry Pi 3 with a similar error message reported in e.g. Debian Bug #821400 or at https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=138629&start=25:
brcmfmac_sdio mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430-sdio.bin failed with error -2

After downgrading to version: 043+rpi5 (stable) wlan0 appears and works again.

043+rpi5 (stable) provides the same version of brcmfmac43430-sdio.bin found at https://github.com/RPi-Distro/firmware-nonfree/tree/master/brcm80211/brcm/. 20161130-2 (testing) provides something else that doesn't work on a Raspberry Pi 3.

I wonder if this package could be changed to have wireless working again.

I am using Raspbian GNU/Linux stretch/sid with kernel 4.4.38-v7+ #938 SMP Thu Dec 15 15:22:21 GMT 2016 armv7l

I reported this problem to Debian because the Raspbian packet states:

Maintainer: Debian Kernel Team <debian-kernel@lists.debian.org>

However I was told this might not be correct either. See

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=850380.

XECDesign, 2017-01-06:
To put this issue into context:
We (Raspberry Pi) run archive.raspberrypi.org and support only stable.
Raspbian supports stable, runs raspbian.org and rebuilds from Debian.

Debian did not have brcmfmac43430-sdio.bin, so we provided our own. This also required adding brcmfmac43430-sdio.txt/

Debian have added the .bin in testing, but are missing the .txt, which might be why it doesn't work.
You can try by pulling it from here https://github.com/RPi-Distro/firmware-nonfree/raw/master/brcm80211/brcm/brcmfmac43430-sdio.txt

If that works, it may be worth bumping the issue with Debian again.

fkbreitl, 2017-01-06:
The latest version of brcmfmac43430-sdio.bin and brcmfmac43430-sdio.txt are found at https://github.com/RPi-Distro/firmware-nonfree/tree/master/brcm80211/brcm/. They are already included in jessie and work well.

I don't know where these two files in testing came from, but they don't work and therefore should be replaced by the same (latest) version found in jessie or the link above.

XECDesign, 2017-01-06:
The testing version of the package comes from Raspbian and, in turn, Debian, not us. Did adding the .txt file give you working wifi with the testing package?

fkbreitl, 2017-01-06:
So who is maintaining the testing repository and where should I report this bug?

Yes, after overwriting these two files in testing with the jessie version everything works as before.

XECDesign, 2017-01-06:
The testing package you've got has come from Raspbian (https://www.raspbian.org/RaspbianBugs), but the issue is inherited from Debian.

The Debian package which the Raspbian package is built from is missing the .txt file. I am suggesting you point out that the .txt file is missing in the Debian issue you've raised.

You might have trouble finding somebody to take responsibility for this, but it does trace back to Debian.

fkbreitl, 2017-01-06:
As I understand it, the txt files are even more hardware specific and omitted on purpose. I don't know how they get there. Maybe auto-generated - at least I remember reading it could be copied from /sys/firmware/.

Anyways I get the impression, that Raspbian testing is different from Debian testing and rather something like unstable or experimental with wrong maintainer information and nobody feeling responsible. In this case it is not even clear where to address bug report to and I will have to switch back to stable. I notices a number of other problems too.

XECDesign, 2017-01-06:
Raspbian is Debian recompiled for armv6 with HF support. They take the Debian source package and build it without any modifications. Sometimes modifications are required for the package to work correctly, and they do make those changes, but this is not such a case. The package you are referring to would not support this hardware if you were running Debian either. For the most part, Raspbian testing is exactly the same as Debian testing. That is certainly the case for this package.

There is nobody to take responsibility because by moving to testing you've opt out of the work Raspberry Pi have done on top of stock Raspbian.

fkbreitl, 2017-01-09:
Ok, I see. I am just surprised because I thought this work would happen on testing and not on stable.

closed 2017-01-10 by XECDesign

rumpelsepp, 2017-05-04:
Are there any known efforts upstream to fix this? I ran into this just moments ago and hopefully I found the solution here by manually deploying that .txt file. The original debian bugreport seems archived/closed.