RPi-Distro/repo #52: Package apt-transport-https should be installed by default

ViToni, 2017-01-08:
Trying to update with an HTTPS repository enabled I got the following error message:

    root@RasPi:~# apt update
    E: The method driver /usr/lib/apt/methods/https could not be found.
    N: Is the package apt-transport-https installed?

After:

    apt install apt-transport-https

the update worked as normal. I would suggest to enable HTTPS by default and adding  `apt-transport-https` to the standard image.

(I am running `Debian 8 ("Jessie")` which was upgraded from `Debian 7 ("Wheezy")`.)

XECDesign, 2017-01-09:
This wasn't pre-installed because we don't use https by default, but I suppose it makes sense to include it for people who don't want to use http to install it.

ViToni, 2017-01-11:
The packages I wanted to install were only available via HTTPS so this issue was raised by the repositoring NOT being provided via HTTP. Not necessarily depending on user preferences.

closed 2017-01-12 by XECDesign