RPi-Distro/repo #56: kernel header always missing / can't compile modules

zoff99, 2017-01-31:
https://github.com/raspberrypi/linux/issues/1820

XECDesign, 2017-01-31:
We ship a newer kernel than what most of debian's dkms packages can build against.
```
DKMS make.log for v4l2loopback-0.8.0 for kernel 4.4.38-v7+ (armv7l)
Tue 31 Jan 13:23:55 UTC 2017
Building v4l2-loopback driver...
make -C /lib/modules/4.4.38-v7+/build M=/var/lib/dkms/v4l2loopback/0.8.0/build modules
make[1]: Entering directory '/usr/src/linux-headers-4.4.38-v7+'
  CC [M]  /var/lib/dkms/v4l2loopback/0.8.0/build/v4l2loopback.o
/var/lib/dkms/v4l2loopback/0.8.0/build/v4l2loopback.c: In function ‘attr_store_maxopeners’:
/var/lib/dkms/v4l2loopback/0.8.0/build/v4l2loopback.c:486:2: error: implicit declaration of function ‘strict_strtoul’ [-Werror=implicit-function-declaration]
  if (strict_strtoul(buf, 0, &curr))
  ^
/var/lib/dkms/v4l2loopback/0.8.0/build/v4l2loopback.c: In function ‘init_vdev’:
/var/lib/dkms/v4l2loopback/0.8.0/build/v4l2loopback.c:2002:7: error: ‘struct video_device’ has no member named ‘debug’
   vdev->debug = V4L2_DEBUG_IOCTL | V4L2_DEBUG_IOCTL_ARG;
       ^
/var/lib/dkms/v4l2loopback/0.8.0/build/v4l2loopback.c:2002:17: error: ‘V4L2_DEBUG_IOCTL’ undeclared (first use in this function)
   vdev->debug = V4L2_DEBUG_IOCTL | V4L2_DEBUG_IOCTL_ARG;
                 ^
/var/lib/dkms/v4l2loopback/0.8.0/build/v4l2loopback.c:2002:17: note: each undeclared identifier is reported only once for each function it appears in
/var/lib/dkms/v4l2loopback/0.8.0/build/v4l2loopback.c:2002:36: error: ‘V4L2_DEBUG_IOCTL_ARG’ undeclared (first use in this function)
   vdev->debug = V4L2_DEBUG_IOCTL | V4L2_DEBUG_IOCTL_ARG;
                                    ^
cc1: some warnings being treated as errors
scripts/Makefile.build:264: recipe for target '/var/lib/dkms/v4l2loopback/0.8.0/build/v4l2loopback.o' failed
make[2]: *** [/var/lib/dkms/v4l2loopback/0.8.0/build/v4l2loopback.o] Error 1
Makefile:1402: recipe for target '_module_/var/lib/dkms/v4l2loopback/0.8.0/build' failed
make[1]: *** [_module_/var/lib/dkms/v4l2loopback/0.8.0/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.4.38-v7+'
Makefile:42: recipe for target 'v4l2loopback.ko' failed
make: *** [v4l2loopback.ko] Error 2
```

Not something we can do much about.

zoff99, 2017-01-31:
alright. but where can i download the correct kernel source?
i can make the link myself.
apt-get does not list sources for the 4.4.44-v7+ kernel.
can you always add a package for kernel sources to install please?

XECDesign, 2017-01-31:
The kernel source is corrext. The v4l2loopback source isn't.
Raspberrypi-kernel-headers is enough to build a module, but if you want the full source of the kernel found in the current raspberrypi-kernel package, you can get the source from https://github.com/raspberrypi/linux/archive/raspberrypi-kernel_1.20161215-1.tar.gz

zoff99, 2017-01-31:
i have downloaded the v4l2loopback also from gitub.

there are just no headers in /usr/lib/modules/<current kernel version>
and no apt-get package to install them

XECDesign, 2017-01-31:
```
$ apt-cache policy raspberrypi-kernel-headers
raspberrypi-kernel-headers:
  Installed: 1.20161215-1
  Candidate: 1.20161215-1
  Version table:
 *** 1.20161215-1 0
        500 http://archive.raspberrypi.org/debian/ jessie/main armhf Packages
        100 /var/lib/dpkg/status
```
Do you see something different?

zoff99, 2017-01-31:
yes but nothing installs into /lib/modules/... for the current kernel version.
raspberrypi-kernel-headers are too old

XECDesign, 2017-01-31:
What do you mean by 'current'? The current kernel in raspberrypi-kernel is 4.4.38.
```
$ ls -l /lib/modules/$(uname -r)/build
lrwxrwxrwx 1 root root 33 Dec 19 20:19 /lib/modules/4.4.38-v7+/build -> /usr/src/linux-headers-4.4.38-v7+
```
If you're getting a pre-built kernel from somewhere else, I'd know nothing about that. If you're building your own kernel, then you already have the source and can use that to build v4l2loopback.

zoff99, 2017-01-31:
well if you do rpi update then it is 4.4.44-v7+
and before it was 4.4.37 which was missing header as well

XECDesign, 2017-01-31:
rpi-update grabs the kernel and firmware from https://github.com/Hexxeh/rpi-firmware/, which follows raspberrypi/firmware. You can the kernel git hash for that here https://github.com/raspberrypi/firmware/blob/master/extra/git_hash

If you had a kernel older that 4.4.38, then a `sudo apt-get update && sudo apt-get dist-upgrade && sudo apt-get install raspberrypi-kernel-headers` would've brought you up to date and made sure you had the kernel headers.

Note that dpkg doesn't handle packages with many files particularly quickly, so installs and upgrades of raspberrypi-kernel-headers would take a very long time. In many cases, it's not the best option.

zoff99, 2017-01-31:
i see.
is there an easy way to get the raspberrypi-kernel-headers for my current kernel?
a command that will work the next time also?

XECDesign, 2017-01-31:
No, unfortunately not. rpi-update is completely decoupled from apt. You might want to take a look at https://github.com/notro/rpi-source/wiki, but that will get you the whole source rather than headers only (not that there is anything wrong with that).

zoff99, 2017-01-31:
so next time only apt-get distupgrade then? but will that upgrade firmware also?

XECDesign, 2017-01-31:
yes, the firmware comes from the raspberrypi-bootloader package, which is updated at the same time as raspberrypi-kernel and raspberrypi-kernel-headers.

closed 2017-04-04 by XECDesign