RPi-Distro/repo #58: Chromium crashes when started over X11 forwarding

moelski, 2017-03-08:
Hi !

With the latest update I can´t longer start Chromium over SSH via x11 forwarding. I always get this error: 
```
pi@raspberrypi:~ $ chromium-browser
bootstrap_helper: /usr/lib/chromium-browser/nacl_helper: Cannot open ELF file!  errno=2
[1:1:0308/065907.001911:ERROR:nacl_fork_delegate_linux.cc(315)] Bad NaCl helper startup ack (0 bytes)
[2958:2958:0308/065908.476292:ERROR:sandbox_linux.cc(343)] InitializeSandbox() called with multiple threads in process gpu-process.
[2902:2944:0308/065908.499507:ERROR:browser_gpu_channel_host_factory.cc(125)] Failed to create channel.
[2902:2944:0308/065908.499916:ERROR:browser_gpu_channel_host_factory.cc(125)] Failed to create channel.
[2902:2944:0308/065908.500097:ERROR:browser_gpu_channel_host_factory.cc(125)] Failed to create channel.
[2902:2944:0308/065908.500221:ERROR:browser_gpu_channel_host_factory.cc(125)] Failed to create channel.
[2902:2944:0308/065908.500335:ERROR:browser_gpu_channel_host_factory.cc(125)] Failed to create channel.
Segmentation fault
```
And sometimes I got this: 
```
pi@raspberrypi:~ $ chromium-browser
bootstrap_helper: /usr/lib/chromium-browser/nacl_helper: Cannot open ELF file!  errno=2
[1:1:0308/065148.085856:ERROR:nacl_fork_delegate_linux.cc(315)] Bad NaCl helper startup ack (0 bytes)
[2873:2873:0308/065150.693624:ERROR:sandbox_linux.cc(343)] InitializeSandbox() called with multiple threads in process gpu-process.
[7:7:0100/000000.635524:ERROR:broker_posix.cc(41)] Invalid node channel message
Segmentation fault
```

Chromium is the latest version : 56.0.2924.84-0ubuntu0.14.04.1.1000
Kernel is : Linux raspberrypi 4.4.50-v7+ #970 SMP Mon Feb 20 19:18:29 GMT 2017 armv7l GNU/Linux

I´ve double checked it and it worked fine before the last update. This problem also exists in the new Raspbian Image (2017-03-02-raspbian-jessie). 

Would be great if you can fix this issue. Let me know if you need any further information. 

Dominik

XECDesign, 2017-03-08:
Looks okay here. Over SSH:
```
pi@raspberrypi:~ $ chromium-browser 
bootstrap_helper: /usr/lib/chromium-browser/nacl_helper: Cannot open ELF file!  errno=2
[1:1:0308/141449.959670:ERROR:nacl_fork_delegate_linux.cc(315)] Bad NaCl helper startup ack (0 bytes)
[18943:18943:0308/141450.920508:ERROR:shared_image_x11.cc(129)] SHM attach failed
[19042:19042:0308/141453.134318:ERROR:sandbox_linux.cc(343)] InitializeSandbox() called with multiple threads in process gpu-process. 
[18943:19258:0308/141521.390862:ERROR:object_proxy.cc(583)] Failed to call method: org.freedesktop.UPower.GetDisplayDevice: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
[18943:19258:0308/141521.392261:ERROR:object_proxy.cc(583)] Failed to call method: org.freedesktop.UPower.EnumerateDevices: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
```

Exits cleanly after browsing around a bit and then closing the window.

```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.4.50-v7+ #970 SMP Mon Feb 20 19:18:29 GMT 2017 armv7l GNU/Linux
pi@raspberrypi:~ $ sudo vcgencmd version
Mar  3 2017 13:42:34 
Copyright (c) 2012 Broadcom
version 9ae30f71c7ef4239e9d5b56346c0842f3ef56736 (clean) (release)
pi@raspberrypi:~ $ apt-cache policy chromium-browser 
chromium-browser:
  Installed: 56.0.2924.84-0ubuntu0.14.04.1.1000
  Candidate: 56.0.2924.84-0ubuntu0.14.04.1.1000
  Version table:
 *** 56.0.2924.84-0ubuntu0.14.04.1.1000 0
        500 http://archive.raspberrypi.org/debian/ jessie/main armhf Packages
        100 /var/lib/dpkg/status
pi@raspberrypi:~ $ cat /proc/device-tree/model 
Raspberry Pi 3 Model B Rev 1.2
```

I will forward this on though.

moelski, 2017-03-09:
Hi !

Thx for your reply. 

I just give it another try on a Raspberry Pi Zero. Before I execute an dist-upgrade it works: 
```
pi@homePiZero:~ $ uname -a
Linux homePiZero 4.4.38+ #938 Thu Dec 15 15:17:54 GMT 2016 armv6l GNU/Linux
pi@homePiZero:~ $ sudo vcgencmd version
Dec  9 2016 15:11:26
Copyright (c) 2012 Broadcom
version 2e557d8dac70add28597c3b449cb52c34588d818 (clean) (release)
pi@homePiZero:~ $ apt-cache policy chromium-browser
chromium-browser:
  Installed: 51.0.2704.91-0ubuntu0.14.04.1.7015
  Candidate: 56.0.2924.84-0ubuntu0.14.04.1.1000
  Version table:
     56.0.2924.84-0ubuntu0.14.04.1.1000 0
        500 http://archive.raspberrypi.org/debian/ jessie/main armhf Packages
 *** 51.0.2704.91-0ubuntu0.14.04.1.7015 0
        100 /var/lib/dpkg/status
pi@homePiZero:~ $ chromium-browser
Received signal 11 SEGV_MAPERR 000000000094
#0 0x00007fdfe098 base::debug::StackTrace::StackTrace()
#1 0x00007fdfdfe8 base::debug::(anonymous namespace)::StackDumpSignalHandler()
#2 0x0000b5b7e1a0 <unknown>
[end of stack trace]
pi@homePiZero:~ $ cat /proc/device-tree/model
Raspberry Pi Zero Rev 1.3
```

Und after update / dist-upgrade I got this: 
```
pi@homePiZero:~ $ uname -a
Linux homePiZero 4.4.50+ #970 Mon Feb 20 19:12:50 GMT 2017 armv6l GNU/Linux
pi@homePiZero:~ $ sudo vcgencmd version
Mar  3 2017 13:42:34
Copyright (c) 2012 Broadcom
version 9ae30f71c7ef4239e9d5b56346c0842f3ef56736 (clean) (release)
pi@homePiZero:~ $ apt-cache policy chromium-browser
chromium-browser:
  Installed: 56.0.2924.84-0ubuntu0.14.04.1.1000
  Candidate: 56.0.2924.84-0ubuntu0.14.04.1.1000
  Version table:
 *** 56.0.2924.84-0ubuntu0.14.04.1.1000 0
        500 http://archive.raspberrypi.org/debian/ jessie/main armhf Packages
        100 /var/lib/dpkg/status
pi@homePiZero:~ $ chromium-browser
bootstrap_helper: /usr/lib/chromium-browser/nacl_helper: Cannot open ELF file!  errno=2
[1:1:0309/052045.411394:ERROR:nacl_fork_delegate_linux.cc(315)] Bad NaCl helper startup ack (0 bytes)
Segmentation fault
pi@homePiZero:~ $
```

And in some cases the error differs a little bit: 
```
pi@homePiZero:~ $ chromium-browser
bootstrap_helper: /usr/lib/chromium-browser/nacl_helper: Cannot open ELF file!  errno=2
[1:1:0309/052133.338289:ERROR:nacl_fork_delegate_linux.cc(315)] Bad NaCl helper startup ack (0 bytes)
[9:9:0100/000000.746057:ERROR:broker_posix.cc(41)] Invalid node channel message
[7:7:0100/000000.744285:ERROR:broker_posix.cc(41)] Invalid node channel message
Segmentation fault
```

Very strange that it works for you. Can I do some additional tests to figure out the problem? 
Which kind of X11 Server did you use for the forwarding? 

regards Dominik

XECDesign, 2017-03-09:
Standard xorg xserver from Ubuntu xenial:
```
Installed: 2:1.18.4-0ubuntu0.2
```

Tried this on a pi zero and it still works.

koopa, 2017-03-10:
Facing the same issue here:
```
Starting program: /usr/lib/chromium-browser/chromium-browser --disable-quic --enable-fast-unload --enable-tcp-fast-open  --enable-pinch --ppapi-flash-path=/usr/lib/chromium-browser/libpepflashplayer.so --ppapi-flash-args=enable_stagevideo_auto=0
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
[New Thread 0x744c1290 (LWP 21267)]
bootstrap_helper: /usr/lib/chromium-browser/nacl_helper: Cannot open ELF file!  errno=2
[1:1:0310/210113.435572:ERROR:nacl_fork_delegate_linux.cc(315)] Bad NaCl helper startup ack (0 bytes)
[New Thread 0x73aff290 (LWP 21275)]
[New Thread 0x730ff290 (LWP 21276)]
[New Thread 0x727ff290 (LWP 21277)]
[New Thread 0x71dff290 (LWP 21278)]
[New Thread 0x715ff290 (LWP 21279)]
[New Thread 0x70dff290 (LWP 21280)]
[New Thread 0x73c08290 (LWP 21281)]
[New Thread 0x6ed37290 (LWP 21282)]
[New Thread 0x6e537290 (LWP 21283)]
[New Thread 0x6dd37290 (LWP 21284)]
[New Thread 0x6d537290 (LWP 21285)]
[New Thread 0x6cd37290 (LWP 21286)]
[New Thread 0x6c537290 (LWP 21287)]
[New Thread 0x6bd37290 (LWP 21288)]
[New Thread 0x6b537290 (LWP 21289)]
[New Thread 0x6ad37290 (LWP 21290)]
[New Thread 0x6a537290 (LWP 21291)]
[New Thread 0x69806290 (LWP 21292)]
[Thread 0x69806290 (LWP 21292) exited]
[New Thread 0x69806290 (LWP 21302)]
[New Thread 0x68eaa290 (LWP 21303)]
[New Thread 0x684ff290 (LWP 21304)]
[New Thread 0x67cff290 (LWP 21305)]
[New Thread 0x671de290 (LWP 21306)]
[New Thread 0x669de290 (LWP 21307)]
[New Thread 0x65fff290 (LWP 21308)]
[Thread 0x65fff290 (LWP 21308) exited]
[New Thread 0x6569f290 (LWP 21310)]
[New Thread 0x64e9f290 (LWP 21313)]
[Thread 0x64e9f290 (LWP 21313) exited]
[New Thread 0x64e9f290 (LWP 21318)]
[New Thread 0x6469f290 (LWP 21321)]
[Thread 0x6469f290 (LWP 21321) exited]

Program received signal SIGSEGV, Segmentation fault.
0x76708224 in xcb_setup_vendor_end () from /usr/lib/arm-linux-gnueabihf/libxcb.so.1
(gdb) s
Single stepping until exit from function xcb_setup_vendor_end,
which has no line number information.
[Thread 0x6569f290 (LWP 21310) exited]
[Thread 0x64e9f290 (LWP 21318) exited]
[Thread 0x669de290 (LWP 21307) exited]
[Thread 0x671de290 (LWP 21306) exited]
[Thread 0x67cff290 (LWP 21305) exited]
[Thread 0x684ff290 (LWP 21304) exited]
[Thread 0x68eaa290 (LWP 21303) exited]
[Thread 0x69806290 (LWP 21302) exited]
[Thread 0x6a537290 (LWP 21291) exited]
[Thread 0x6ad37290 (LWP 21290) exited]
[Thread 0x6b537290 (LWP 21289) exited]
[Thread 0x6c537290 (LWP 21287) exited]
[Thread 0x6cd37290 (LWP 21286) exited]
[Thread 0x6d537290 (LWP 21285) exited]
[Thread 0x6dd37290 (LWP 21284) exited]
[Thread 0x6e537290 (LWP 21283) exited]
[Thread 0x6ed37290 (LWP 21282) exited]
[Thread 0x73c08290 (LWP 21281) exited]
[Thread 0x70dff290 (LWP 21280) exited]
[Thread 0x715ff290 (LWP 21279) exited]
[Thread 0x71dff290 (LWP 21278) exited]
[Thread 0x727ff290 (LWP 21277) exited]
[Thread 0x730ff290 (LWP 21276) exited]
[Thread 0x73aff290 (LWP 21275) exited]
[Thread 0x744c1290 (LWP 21267) exited]
[Thread 0x7512c000 (LWP 21223) exited]
[9:9:0100/000000.782471:ERROR:broker_posix.cc(41)] Invalid node channel message
[7:7:0100/000000.782572:ERROR:broker_posix.cc(41)] Invalid node channel message

Program terminated with signal SIGSEGV, Segmentation fault.
The program no longer exists.
```

moelski, 2017-03-11:
Hi koopa, 
great - I´m not alone with this problem :-)

Which SSH tool did you use? I use MobaXTerm on Windows and XECDesign uses the standart from ubuntu.

koopa, 2017-03-11:
BTW the backtrace after the segfault:
```
#0  xcb_setup_vendor_end (R=0x0) at xproto.c:1569
#1  0x7670725c in xcb_setup_pixmap_formats (R=<optimized out>) at xproto.c:1588
#2  0x766e6fac in ?? () from /usr/lib/arm-linux-gnueabihf/libxcb-image.so.0
```
@moelski I was using Xming and putty on Windows 10.
Earlier I also tried forwarding to a mac and it worked (using XQuartz).

ydczbyq, 2017-04-02:
I have the same problem and need to start chromium in kiosk mode on boot.  It was working fine for a couple hours then I started receiving the same error as others in this thread.  Please help!

serghix, 2017-04-25:
Hi all,

I am currently using Xming and PuTTy on Windows 7 to connect to my RPi. I had no problems with the old Raspian OS, however, when I installed the updated Raspian with PIXEL on my RPi, Chromium stopped working.

When I click to open the browser, it tries to open, then shuts down. From Terminal, I get the following error messages:

pi@raspberrypi:~ $ chromium-browser
bootstrap_helper: /usr/lib/chromium-browser/nacl_helper: Cannot open ELF file! errno=2
[1:1:0425/092537.790686:ERROR:nacl_fork_delegate_linux.cc(315)] Bad NaCl helper startup ack (0 bytes)
[1382:1382:0425/092549.681335:ERROR:gl_surface_glx.cc(411)] GLX 1.3 or later is required.
[1382:1382:0425/092549.682982:ERROR:gl_initializer_x11.cc(130)] GLSurfaceGLX::InitializeOneOff failed.
[1382:1382:0425/092549.762006:ERROR:gpu_child_thread.cc(327)] Exiting GPU process due to errors during initialization
[1331:1373:0425/092549.850833:ERROR:browser_gpu_channel_host_factory.cc(113)] Failed to launch GPU process.
[1331:1373:0425/092549.870017:ERROR:browser_gpu_channel_host_factory.cc(113)] Failed to launch GPU process.
[1331:1373:0425/092549.875323:ERROR:browser_gpu_channel_host_factory.cc(113)] Failed to launch GPU process.
[1331:1373:0425/092549.884174:ERROR:browser_gpu_channel_host_factory.cc(113)] Failed to launch GPU process.
[1331:1373:0425/092549.887169:ERROR:browser_gpu_channel_host_factory.cc(113)] Failed to launch GPU process.
Segmentation fault

Any suggestions would be really helpful, because I have tried everything.

Thanks.

wvxvw, 2017-05-12:
Same problem exists on Raspbian:

``` sh
$ ssh -X pi@10.0.0.19
$ uname -a
Linux raspberrypi 4.9.24-v7+ #993 SMP Wed Apr 26 18:01:23 BST 2017 armv7l GNU/Linux

$ chromium-browser 
bootstrap_helper: /usr/lib/chromium-browser/nacl_helper: Cannot open ELF file!  errno=2
[1:1:0512/070732.197629:ERROR:nacl_fork_delegate_linux.cc(315)] Bad NaCl helper startup ack (0 bytes)
Xlib:  extension "RANDR" missing on display "localhost:10.0".
Xlib:  extension "XInputExtension" missing on display "localhost:10.0".
Xlib:  extension "RANDR" missing on display "localhost:10.0".
Xlib:  extension "XInputExtension" missing on display "localhost:10.0".
Xlib:  extension "XInputExtension" missing on display "localhost:10.0".
Xlib:  extension "SHAPE" missing on display "localhost:10.0".
Xlib:  extension "SHAPE" missing on display "localhost:10.0".
Xlib:  extension "SHAPE" missing on display "localhost:10.0".
Xlib:  extension "SHAPE" missing on display "localhost:10.0".
Xlib:  extension "SHAPE" missing on display "localhost:10.0".
[1:1:0100/000000.816196:ERROR:broker_posix.cc(41)] Invalid node channel message
```

wvxvw, 2017-05-12:
UPDATE: In my case, I found the problem: I had to add `-Y` option to SSH to make it work:

``` sh
ssh -YX pi@10.0.0.9 chromium-browser
```

works (my SSH client is on Arch Linux)

``` sh
$ ssh -V
OpenSSH_7.5p1, OpenSSL 1.1.0e  16 Feb 2017
```

kriegerkl, 2017-05-16:
For me ssh -YX .. don't work.
I get the same error:
bootstrap_helper: /usr/lib/chromium-browser/nacl_helper: Cannot open ELF file!  errno=2
[1:1:0516/221253.723885:ERROR:nacl_fork_delegate_linux.cc(315)] Bad NaCl helper startup ack (0 bytes)
[16790:16790:0516/221253.760578:ERROR:browser_main_loop.cc(495)] Failed to put Xlib into threaded mode.
[16790:16790:0516/221253.781850:ERROR:browser_main_loop.cc(272)] Gtk: cannot open display:

Any suggestions would be really helpful, because I have tried everything.

Thanks.

bryan-dickson, 2017-05-16:
bootstrap_helper: /usr/lib/chromium-browser/nacl_helper: Cannot open ELF file!  errno=2
[1:1:0516/231546.290034:ERROR:nacl_fork_delegate_linux.cc(315)] Bad NaCl helper startup ack (0 bytes)
[3975:3975:0516/231546.707183:ERROR:browser_main_loop.cc(495)] Failed to put Xlib into threaded mode.
[3975:3975:0516/231546.820477:ERROR:browser_main_loop.cc(272)] Gtk: cannot open display:

koopa, 2017-05-18:
@bryan-dickson are you sure X forwarding is enabled and your `$DISPLAY` is set correctly?

bryan-dickson, 2017-05-22:
Got the window to open with this, but still seeing an error:

```
pi@brypi:~ $ /usr/bin/chromium-browser
bootstrap_helper: /usr/lib/chromium-browser/nacl_helper: Cannot open ELF file!  errno=2
[1:1:0522/221001.431301:ERROR:nacl_fork_delegate_linux.cc(315)] Bad NaCl helper startup ack (0 bytes)
[17205:17205:0522/221001.870030:ERROR:browser_main_loop.cc(495)] Failed to put Xlib into threaded mode.
[17205:17205:0522/221002.054090:ERROR:browser_main_loop.cc(272)] Gtk: cannot open display: 
pi@brypi:~ $ echo $DISPLAY

pi@brypi:~ $ export DISPLAY=:0
pi@brypi:~ $ echo $DISPLAY
:0
pi@brypi:~ $ /usr/bin/chromium-browser
bootstrap_helper: /usr/lib/chromium-browser/nacl_helper: Cannot open ELF file!  errno=2
[1:1:0522/221049.348795:ERROR:nacl_fork_delegate_linux.cc(315)] Bad NaCl helper startup ack (0 bytes)
Created new window in existing browser session.
pi@brypi:~ $ 
```

moelski, 2017-05-26:
@bryan-dickson : DISPLAY 0 is the attached Display. 
And it looks like there was an instance of chromium opened: 
> "Created new window in existing browser session."

So I don´t think you have the same issue as described in my first post ...

9swampy, 2017-07-06:
Adding my 2 cents worth...

Brand new Pi 3 with Raspian, installed XRdp and remoting in via Win7 RDP. All else so far seems stable but within a 15 seconds of starting Chromium 56.0.2924.84-0ubuntu0.14.04.1.1000 every time the RDP session terminates unless I very quickly quit back out of Chromium after starting.

> pi@raspberrypi:~ $ chromium-browser
> bootstrap_helper: /usr/lib/chromium-browser/nacl_helper: Cannot open ELF file!  errno=2
> [1:1:0706/210453.353142:ERROR:nacl_fork_delegate_linux.cc(315)] Bad NaCl helper startup ack (0 bytes)
> Xlib:  extension "RANDR" missing on display ":10.0".
> Xlib:  extension "XInputExtension" missing on display ":10.0".
> Xlib:  extension "RANDR" missing on display ":10.0".
> Xlib:  extension "XInputExtension" missing on display ":10.0".
> Xlib:  extension "XInputExtension" missing on display ":10.0".
> Xlib:  extension "XInputExtension" missing on display ":10.0".
> [1774:1774:0706/210455.692980:ERROR:gl_surface_glx.cc(406)] glxQueryVersion failed
> [1774:1774:0706/210455.693587:ERROR:gl_initializer_x11.cc(130)] GLSurfaceGLX::InitializeOneOff failed.
> [1774:1774:0706/210455.773614:ERROR:gpu_child_thread.cc(327)] Exiting GPU process due to errors during initialization
> [1727:1765:0706/210457.578249:ERROR:browser_gpu_channel_host_factory.cc(113)] Failed to launch GPU process.
> [1727:1765:0706/210457.671602:ERROR:browser_gpu_channel_host_factory.cc(113)] Failed to launch GPU process.
> [1727:1765:0706/210457.673284:ERROR:browser_gpu_channel_host_factory.cc(113)] Failed to launch GPU process.
> [1727:1765:0706/210457.674121:ERROR:browser_gpu_channel_host_factory.cc(113)] Failed to launch GPU process.
> [1727:1765:0706/210457.674499:ERROR:browser_gpu_channel_host_factory.cc(113)] Failed to launch GPU process.
> [1727:1765:0706/210457.674835:ERROR:browser_gpu_channel_host_factory.cc(113)] Failed to launch GPU process.
> [1727:1727:0706/210457.676299:ERROR:shared_image_x11.cc(129)] SHM attach failed
> pi@raspberrypi:~ $
>

XECDesign, 2017-08-31:
Has this been resolved with the new version of Chromium (in stretch)?

moelski, 2017-08-31:
Hi, 

for me it works. 

Dominik

closed 2017-08-31 by XECDesign

morphx666, 2017-12-17:
Why was this issue closed? I'm having the exact same issue using Bitvise and Xming.

XECDesign, 2017-12-17:
@morphx666 
What's the output of `apt-cache policy chromium-browser`?

morphx666, 2017-12-17:
```
pi@raspberrypi:~ $ apt-cache policy chromium-browser
chromium-browser:
  Installed: 56.0.2924.84-0ubuntu0.14.04.1.1000
  Candidate: 56.0.2924.84-0ubuntu0.14.04.1.1000
  Version table:
 *** 56.0.2924.84-0ubuntu0.14.04.1.1000 0
        500 http://archive.raspberrypi.org/debian/ jessie/main armhf Packages
        100 /var/lib/dpkg/status
```

XECDesign, 2017-12-17:
Looks like you're running Jessie. Stretch is the currently supported release.
https://www.raspberrypi.org/blog/raspbian-stretch/