RPi-Distro/repo #65: able to crash Raspberry Pi X server

woodlloydwood, 2017-04-18:
Occasional Raspberry Pi model B user here. Just upgraded to PIXEL 1.2.

Discovered that the packages I maintain when run together now cause the Pi X server to freeze. (savi and geomview, just:
sudo apt-get install geomview
sudo apt-get install savi
geomview -run savi
)

This appears to be a gradual decrease in functionality over time with recent Pi distro upgrades; geomview used to be able to render texturemapped spheres just fine, then it could only render half of a texturemapped Earth sphere but still work, now with PIXEL geomview attempting to draw just freezes the X server for all applications and locks everything up. Which appears to be a bad thing.

_mgopengl_ctxset: undefined option: 1596824

is the error; I don't see this behaviour on other platforms (x86_64 Linux, mostly, some Mac.) Where to report this to whoever has an interest in maintaining X on the Pi?

http://savi.sf.net/
http://www.geomview.org/

thanks.

XECDesign, 2017-04-18:
If you're using the experimental, GL driver, it's best to report the issue here https://github.com/anholt/mesa/issues.

Although I would recommend providing more information there. I've just tried it with the fkms driver and it seems to be working.

woodlloydwood, 2017-04-22:
This is an original Raspberry Pi B, using stock software mesa rendering. What rendering there is is in monochrome, including glxgears, which gets about 0.1 frames a second. Experimental or hardware rendering is not supported on model 0 or 1, so it seems mesa is all there is. fkms does not appear to be an option.

It appears that no-one is testing on the original model B; for example, PIXEL replaces iceweasel and installs firefox-esr, which is painfully painfully unusably slow.

XECDesign, 2017-04-24:
We have never shipped with iceweasel or firefox-esr. Debian have switched the branding of iceweasel back to firefox, since the legal issues surrounding it have been resolved. If the newer version of firefox is slower, that's entirely due to what's happening upstream in Debian and Firefox development.

We have shipped with, and continue to ship with Epiphany and have recently added Chromium as the default option.

Is the 0.1 frames per second about what you expect with software rendering? I believe that the software rendering libraries have not been modified, but have been updated to a newer version. I think it's the version that will be in Debian Stretch. Re-opening to take a look at this later.

woodlloydwood, 2017-04-25:
(I should have said 'flat solid colour' for monochrome. no light sourcing.)

Thanks for followng up.

This little 256MB Model B is far more powerful than the hardware SaVI and Geomview started out on over twenty years ago, so I would not expect problems with OpenGL rendering even is software at this point. They're not massive CPU users; render a sphere and that's pretty much it.
Compare the output of 
geomview -run savi
geomview -noopengl -run savi
to compare OpenGL and X rendering. Obviously you'd want to disable hardware rendering on your newer hardware... it's not an option that can be turned on in config on the B

firefox-esr just about works if I decrease graphics ram. A tight fit memorywise, presumably paging massively when in decreased space.

Epiphany is installed and on path - aha, thanks. How to install Chromium? sudo apt-get chromium does not do what you might expect; it is neither chromium-inspector nor chromium-bsu, which do not replace it.

XECDesign, 2017-04-25:
It should just be chromium-browser? Is this a very old image? Do you have the ui component enabled in /etc/apt/sources.list.d/raspi.list?
On a 256MB model, I would expect epiphany to perform better, since Chromium can eat up quite a bit of RAM.

woodlloydwood, 2017-04-25:
Thanks, got chromium-browser installed. Odd that it isn't suggested by apt-get. Interesting too that sound works in Chromium tabs (once the sound module is installed in PIXEL 1.2, natch), but not in other browsers. (Was chromium added after the Feb PIXEL 1.2 release?)

But this is getting a little away from the OpenGL rendering issue.

XECDesign, 2017-04-25:
Don't know about PIXEL version numbers, but it was included in the 2016-09-23 image.

XECDesign, 2017-09-04:
Not much to report on this. It looks like the change in behaviour comes from upstream Mesa and/or Xorg. I think it would be good for the geomview devs to take a look at why it's not behaving, but the last release seems to have been in 2014. I can't pin this on anything that we've done.

XECDesign, 2017-09-05:
Ah, I see this has already been reported in a more appropriate place:
https://bugs.freedesktop.org/show_bug.cgi?id=101338

closed 2017-09-05 by XECDesign

woodlloydwood, 2017-09-06:
Thanks - I had some considerable difficulty figuring out the correct place
to report this. Raspberry model Bs run the Mesa driver,
not the accelerated driver later models use, it's not
viewed as a Raspbian or PIXEL problem, etc. A bunch of
different mailing systems, repositories, maintainer
groups, and so on.

How anything in Linux ever gets reported to someone who
is resonsible for it, can fix it, and cares enough
to fix it is entirely beyond me!

https://bugs.freedesktop.org/show_bug.cgi?id=101338