RPi-Distro/repo #66: different /etc/machine-id on each restart

andrewmichaelsmith, 2017-05-09:
With a fresh image of Raspbian Jessie Lite 2017-04-10 I find the machine-id changes on each boot.

I've tried this with a Pi 2 and a Pi Zero W and notice the same behaviour:

- Boot for first time observe machine-id X
- sudo reboot
- Observe machine id Y

The machine-id changes on each reboot. I expected it stay the same. This becomes a problem especially when you use journald, which depends on the machine-id staying the same.

Here's what the system logs tell me:
```
pi@raspberrypi:~ $ sudo journalctl | grep machine
Apr 10 09:58:36 raspberrypi systemd[1]: Initializing machine ID from random generator.
Apr 10 09:58:36 raspberrypi systemd[1]: Installed transient /etc/machine-id file.
```

And machine-id is actually a tmpfs mount:

```
pi@raspberrypi:~ $ mount | grep machine
tmpfs on /etc/machine-id type tmpfs (ro,mode=755)
```
Anything else I can check?

XECDesign, 2017-05-10:
Should've been fixed by https://github.com/RPi-Distro/pi-gen/commit/6c146a7bb791a8a4b43ac71061994fb6fe54cee5, which will be in the next image.

andrewmichaelsmith, 2017-05-10:
@XECDesign aha great, thanks!

andrewmichaelsmith, 2017-05-18:
I've just built an image from the "dev" branch on https://github.com/RPi-Distro/pi-gen and can reproduce this issue described above.

One slight difference, if I unmount the mount described above I can see the empty machine-id file that has been created by `export-image/04-finalise/01-run.sh`:

```
pi@raspberrypi:~ $ sudo umount /etc/machine-id 
pi@raspberrypi:/etc $ ls -ltr /etc/ | grep machin
-r--r--r-- 1 root root       0 May 18 21:10 machine-id
```

But this is still being mounted over by a transient machine-id.

Note that to build this image I ran the following:

```
touch ./stage3/SKIP ./stage4/SKIP ./stage5/SKIP
echo "IMG_NAME='Raspbian'" > config
sudo ./build-docker.sh
```

andrewmichaelsmith, 2017-05-18:
I have just realised that I can get older logs from `/var/log/syslog` so this isn't a huge issue. But I think it's still slightly unexpected.

XECDesign, 2017-05-19:
Yes, you are right, that commit wouldn't have fixed anything. I will have a think about a proper fix.

XECDesign, 2017-05-19:
Looks like this isn't something that can be fixed properly until stretch.
systemd has gone through a number of approached to tackling this, but they were added in a few versions above what's in jessie, starting with https://github.com/systemd/systemd/commit/3fb394369ccd6787f8f883e0f81dc7f35cb1ec51.

If you rely on a set machine-id, you could `umount /etc/machine-id` and then run `systemd-machine-id-setup`.

A workaround which preserves the current machine-id is mentioned here:
https://unix.stackexchange.com/a/260542

Our of curiosity, what sorts of issues does this cause with journalctl and otherwise?

andrewmichaelsmith, 2017-05-19:
Thanks for looking in to this and the update. 

The problem comes from when trying to look at journald logs from previous boots. If you set up journald to persist logs to disk you can ordinarily see historic logs with previous boots with `journalctl --list-boots`, this will show you the past set of boots it knows about. 

However - it uses machine-id when looking at it's persisted journal logs, so if the machine-id changes on each boot `journalctl --list-boots` always only has the current boot.

I was planning on using journald to track all the logging for a system I was deploying on top of raspbian and found that after boots I couldn't see historic logs, which is what sent me down this rabbit hole.

As I mentioned, I've found a simple workaround for my use case (ignore journalctl) which I'm happy with but I suppose this might bite someone along the line.

XECDesign, 2017-05-19:
Thanks, that's good to know. I've used `journactl -b -1` after enabling persistent storage and that seemed to work, but I haven't tried `--list-boots`.

XECDesign, 2017-07-17:
This is going to be fixed in stretch, which includes this service: https://www.freedesktop.org/software/systemd/man/systemd-machine-id-commit.service.html

closed 2017-08-31 by XECDesign