RPi-Distro/repo #68: valgrind / QEMU crash: SETEND instruction unsupported (disInstr(arm): unhandled instruction: 0xF1010200)

minfrin, 2017-05-20:
When valgrind is used on the Raspberry Pi for any kind of debugging, valgrind crashes as follows:

==7222== 
disInstr(arm): unhandled instruction: 0xF1010200
                 cond=15(0xF) 27:20=16(0x10) 4:4=0 3:0=0(0x0)
==7222== valgrind: Unrecognised instruction at address 0x48636f4.
==7222==    at 0x48636F4: ??? (in /usr/lib/arm-linux-gnueabihf/libarmmem.so)
==7222== Your program just tried to execute an instruction that Valgrind
==7222== did not recognise.  There are two possible reasons for this.
==7222== 1. Your program has a bug and erroneously jumped to a non-code
==7222==    location.  If you are running Memcheck and you just saw a
==7222==    warning about a bad jump, it's probably your program's fault.
==7222== 2. The instruction is legitimate but Valgrind doesn't handle it,
==7222==    i.e. it's Valgrind's fault.  If you think this is the case or
==7222==    you are not sure, please let us know and we'll try to fix it.
==7222== Either way, Valgrind will now raise a SIGILL signal which will
==7222== probably kill your program.
==7222== 
==7222== Process terminating with default action of signal 4 (SIGILL)
==7222==  Illegal opcode at address 0x48636F4
==7222==    at 0x48636F4: ??? (in /usr/lib/arm-linux-gnueabihf/libarmmem.so)
==7222== 
==7222== HEAP SUMMARY:
==7222==     in use at exit: 35,782 bytes in 546 blocks
==7222==   total heap usage: 853 allocs, 307 frees, 55,700 bytes allocated
==7222== 
==7222== LEAK SUMMARY:
==7222==    definitely lost: 0 bytes in 0 blocks
==7222==    indirectly lost: 0 bytes in 0 blocks
==7222==      possibly lost: 856 bytes in 23 blocks
==7222==    still reachable: 34,626 bytes in 520 blocks
==7222==                       of which reachable via heuristic:
==7222==                         newarray           : 832 bytes in 16 blocks
==7222==         suppressed: 0 bytes in 0 blocks
==7222== Rerun with --leak-check=full to see details of leaked memory
==7222== 
==7222== For counts of detected and suppressed errors, rerun with: -v
==7222== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
Illegal instruction

This is caused by the SETEND instruction, which is impractical to support on valgrind due to severe performance implications as described here:

https://bugs.kde.org/show_bug.cgi?id=322935

It appears that a sub-optimal implementation of memcmp (and possibly other functions) has been implemented in the RPi that break instrumentation and debugging tools. For obvious reasons this renders the RPi severely limited as a teaching tool - the ability to debug effectively is a vital skill.

It is reported that the same bug makes it impossible to run Raspbian under QEMU.

As requested in https://bugs.kde.org/show_bug.cgi?id=322935#c24, please remove the hack from Raspbian so that Raspbian can run in virtualised and debugging environments.

XECDesign, 2017-05-21:
This can be disabled by editing `/etc/ld.so.preload` or removing `raspi-copies-and-fills` . However, this won't be done on the version of Raspbian that we ship, because the performance improvements are too great.

I understand that having to make that modification before using qemu may be a little fiddly, but far from impossible.

minfrin, 2017-05-22:
Can you clarify what modification to /etc/ld.so.preload I should do to disable this?

pi@towerofpi6:~ $ cat /etc/ld.so.preload 
/usr/lib/arm-linux-gnueabihf/libarmmem.so

While I totally understand the rationale behind an asm optimised memcpy, when that code breaks the debugger, all you end up with is a faster route to the scene of the crash.

minfrin, 2017-05-22:
Trying to move /etc/ld.so.preload out of the way makes no difference, valgrind still crashes as before.

minfrin, 2017-05-22:
Correction: Trying to move /etc/ld.so.preload out of the way on a totally different machine makes no difference.

Moving it out of the way on the correct machine makes valgrind pass this point.

minfrin, 2017-05-23:
I have found someone to have factored out the setend instruction and replaced it with rev instructions:

https://github.com/rsaxvc/arm-mem/commit/b836e465c2fd0bb006b428abce99e31607072834

XECDesign, 2017-09-21:
The above fix has been merged into raspi-copies-and-fills.

rsaxvc, 2023-07-16:
Do you have a link for that repository?