RPi-Distro/repo #69: Raspbian Jessie Hash Sum mismatch

kaizoku-619, 2017-06-15:
when i run sudo apt-get update i get the following:
Hit http://mirrordirector.raspbian.org jessie InRelease
Hit http://archive.raspberrypi.org jessie InRelease
Hit http://mirrordirector.raspbian.org jessie/main armhf Packages
Get:1 http://archive.raspberrypi.org jessie/main armhf Packages [163 kB]
Hit http://mirrordirector.raspbian.org jessie/contrib armhf Packages
Get:2 http://mirrordirector.raspbian.org jessie/non-free armhf Packages [84.2 kB]
Get:3 http://mirrordirector.raspbian.org jessie/rpi armhf Packages [1,356 B]
Hit http://archive.raspberrypi.org jessie/ui armhf Packages
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en
Ign http://mirrordirector.raspbian.org jessie/main Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/main Translation-en
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en
Ign http://archive.raspberrypi.org jessie/main Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en_GB
Ign http://archive.raspberrypi.org jessie/main Translation-en
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en
Ign http://archive.raspberrypi.org jessie/ui Translation-en_GB
Ign http://archive.raspberrypi.org jessie/ui Translation-en
Fetched 248 kB in 12s (20.4 kB/s)
W: Failed to fetch http://archive.raspberrypi.org/debian/dists/jessie/main/binary-armhf/Packages Hash Sum mismatch

E: Some index files failed to download. They have been ignored, or old ones used instead.

XECDesign, 2017-06-15:
Looks okay from here, but I'll raise it with the hosting company just in case.

XECDesign, 2017-06-19:
Since there have been a few updated to the repo since this was reported, are you still experiencing the issue?

kaizoku-619, 2017-06-19:
actually when I reinstalled Raspbian & updated it the error was gone

XECDesign, 2017-06-19:
Alright, thanks for the update. Okay to close?

kaizoku-619, 2017-06-19:
Yup thank you

closed 2017-06-19 by XECDesign

davidt99, 2017-07-29:
I having the same issue. Fresh install of jessie light using internet installation. @XECDesign, any chance you can check it out?
I tried to find mirrors for debain repository but couldn't find.
Thanks.

XECDesign, 2017-07-31:
@davidt99 Is the problem still occurring? It seems like a temporary issue that happens when the repo is updated, while the mirrors sync up.

davidt99, 2017-07-31:
Yes, it resolved the day after. Thanks.

adwidianjaya, 2018-06-29:
Hi, I also experienced this issue an hour ago till now, although nothing happened a day before. 
If I remove http://archive.raspberrypi.org/debian from /etc/apt/sources.list.d/raspi.list, no error occured. But several packages were unavailable (e.g. no installation candidate). However, hash sum mismatch only occurs for http://archive.raspberrypi.org. Is it caused by update process on repo as @XECDesign say? 
Thank you.