RPi-Distro/repo #70: /usr/bin/mandb: fork failed: Cannot allocate memory

CNMan, 2017-06-24:
pi 2b
2017-06-21-raspbian-jessie-lite
```bash
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.9.28-v7+ #998 SMP Mon May 15 16:55:39 BST 2017 armv7l GNU/Linux
pi@raspberrypi:~ $ free -m
             total       used       free     shared    buffers     cached
Mem:           923        350        572          6         17        289
-/+ buffers/cache:         44        879
Swap:           99          0         99
pi@raspberrypi:~ $ sudo apt-get install dnsutils
Reading package lists... Done
Building dependency tree
Reading state information... Done
Suggested packages:
  rblcheck
The following NEW packages will be installed:
  dnsutils
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 113 kB of archives.
After this operation, 325 kB of additional disk space will be used.
Get:1 http://mirrors.ustc.edu.cn/raspbian/raspbian/ jessie/main dnsutils armhf 1:9.9.5.dfsg-9+deb8u11 [113 kB]
Fetched 113 kB in 0s (755 kB/s)
Selecting previously unselected package dnsutils.
(Reading database ... 31494 files and directories currently installed.)
Preparing to unpack .../dnsutils_1%3a9.9.5.dfsg-9+deb8u11_armhf.deb ...
Unpacking dnsutils (1:9.9.5.dfsg-9+deb8u11) ...
Processing triggers for man-db (2.7.0.2-5) ...
/usr/bin/mandb: fork failed: Cannot allocate memory
Setting up dnsutils (1:9.9.5.dfsg-9+deb8u11) ...
pi@raspberrypi:~ $
```
appear almost every time when install softwares

XECDesign, 2017-06-24:
Thanks.

Somehow the image ended up with manpage cache that mandb struggles with. It can be regenerated with `sudo mandb -pcq`. The problem can be reproduced by running without the `-c` parameter.

Oddly enough a dump of the databases before and after regenerating the cache looks the same, but copying new files over the old files also fixes the problem.

Although this affects both the lite and desktop images, chrooting into those build stages shows that the problem is introduced at a later stage (image export).

For reference:
https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=186683#p1178706
https://github.com/Distrotech/gdbm/blob/master/src/bucket.c#L67

XECDesign, 2017-06-27:
Could you please confirm whether this fixes the problem?
```bash
wget https://github.com/RPi-Distro/repo/files/1104677/man-db_2.7.5-1.bpo8.1_armhf.zip
unzip man-db_2.7.5-1.bpo8.1_armhf.zip
sudo dpkg -i man-db_2.7.5-1.bpo8.1_armhf.deb
```

[man-db_2.7.5-1~bpo8+1_armhf.zip](https://github.com/RPi-Distro/repo/files/1104677/man-db_2.7.5-1.bpo8.1_armhf.zip)

ikizhvatov, 2017-06-27:
Observed the same problem. @XECDesign after applying your fix the message "Cannot allocate memory" does not show up in the printout of `sudo apt-get install dnsutils`. I did not do any further testing.

XECDesign, 2017-06-27:
Thanks. I'll give it a few more days of testing and push the package to the repo if no problems surface.

CNMan, 2017-06-27:
i can't test it because i have updated to raspbian 9.0:)

```
pi@raspberrypi:~ $ cat /etc/apt/sources.list
#deb http://mirrordirector.raspbian.org/raspbian/ jessie main contrib non-free rpi
# Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://archive.raspbian.org/raspbian/ jessie main contrib non-free rpi

deb http://mirrors.ustc.edu.cn/raspbian/raspbian/ stretch main contrib non-free firmware rpi
#deb-src http://mirrors.ustc.edu.cn/raspbian/raspbian/ stretch main contrib non-free firmware rpi
```

XECDesign, 2017-07-03:
Updated man-db is in the repo now.

closed 2017-07-03 by XECDesign