RPi-Distro/repo #73: dnsmasq in stretch built against jessie libraries.

plugwash, 2017-07-16:
dnsmasq in your stretch repo is built against the jessie libraries. It needs rebuilding for the stretch libraries.

Orignal report at https://bugs.launchpad.net/raspbian/+bug/1704621

XECDesign, 2017-07-16:
Thanks for the heads up.

XECDesign, 2017-07-17:
Should be fixed now.

closed 2017-07-17 by XECDesign

bwduncan, 2017-10-06:
Sorry for reopening a long dead bug but this doesn't seem to be fixed in raspbian stretch. Should the bug be bumped in the raspbian bug tracker instead?

benjaminreichDE, 2017-10-06:
@bwduncan I am the orignal bug reporter. My dependencies are okay. Try  "apt-cache depends dnsmasq-base" and paste your result here.

bwduncan, 2017-10-06:
Sorry, my mistake. I had updated /etc/apt/sources.list but not /etc/apt/sources.list.d/raspi.list. Apologies for the noise!