RPi-Distro/repo #74: Watchdog package has flaws since recent updates

paulvee, 2017-07-22:
The watchdog package does not allow a persistent installation across a reboot using the systemV way of doing things in Jessie Lite (2016-03-18) anymore. The installation messages of the package shown on the console are confusing and do not point to solutions. The only way to get it working across a reboot now is to use rc.local or cron. That is a kludge at best, so what is the correct installation procedure now.

Trying to install watchdog the systemd way is also not working. Although the systemd service files for watchdog and wd_keepalive are installed, they do not work without changes and there are no pointers offered to lead to a proper installation.

I wrote a post on the Raspberri Pi forum that details the problems with the messages and the installation issues.

https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=147501&p=973710#

XECDesign, 2017-07-24:
I am not sure I understand. I've installed the watchdog package, enabled the watchdog in config.txt and system.conf and it seems to just work. sysvinit is something that we don't support. It sounds like what's needed is a documentation page on how to enable and configure the watchdog.

paulvee, 2017-07-24:
XECDesign, if you would be so kind to tell me (exactly) what you did, I may be able to write something up on the Forum. At this moment, it's too unclear for most users, as my post shows. 
You can PM me using the link to my post.

XECDesign, 2017-07-24:
I'll admit that it took a bit of fiddling, since I didn't have the watchdog package installed. Also, I am running stretch, so what I say might not be exactly the same on jessie. I'll go through it more thoroughly at some point and report back.

I enabled the watchdog in config.txt (`dtparam=watchdog=on`) and /etc/systemd/system.conf (`RuntimeWatchdogSec=10`). I didn't have the watchdog package installed, so systemd would enable the watchdog, but never kicked it resulting in my pi being stuck in a bootloop. After installing the watchdog package, it seems to work as expected.

paulvee, 2017-07-24:
Thank you, that was quick.

 

The problem I think we "all" seem to have is that there are two watchdog's.

One is the systemd one, that is started the way you just described.

It needs a few tweaks to make that work, and I describe that in my post on the Raspberry Forum.

 

However, the watchdog package is a completely different and separate way of using the watchdog, and these two don't really play well together. You either use one, or the other.

 

The watchdog package is the original package that has been around for ages. It has been "ported" over to work with systemd in Jessie, but again, it's separate and different.

I also tried to explain that in my post.

 

What is needed is clarification between the two, and a good install for the systemd watchdog. 

 

I hope I made myself clear, if not, please peruse my post.

 

If I can help, let me know,

 

Paul

XECDesign, 2017-07-24:
Ah, in light of that information, the reason I was stuck in a bootloop was that I set the `RuntimeWatchdogSec` value too high (5 minutes).  The driver returned an error when systemd tried to change it and systemd didn't kick it frequently enough or at all. After removing the watchdog package, everything still works.

So I think now I understand that your complaint is specifically about the watchdog package, which I haven't actually used. Is there a reason you would want to use that instead of systemd's built-in approach?

paulvee, 2017-07-24:
I don't want to learn more about the package, that is well documented and known.

I want to understand more about the native systemd method(s) and what changed recently and why.

 

As an example, a few updates ago, with the systemd watchdog enables, if you did a poweroff, the system would automatically reboot in the period set in the watchdog.conf file.

That is no longer the case. This is not really helpful for embedded applications.

Again, I think I have all the questions covered in my post.

 

Do you know who we can ask that is an expert on the systemd watchdog?

 

paulv

rodizio1, 2017-07-24:
Not sure if it's related, but about a year ago (Raspbian 2016-something with Kernel 4.4.11) I also couldn't get it to work. Also, there were strange dmesg messages ("watchdog did not stop" or similar) when doing reboot or halt.

Then I removed the watchdog completely (did not compile it in at all) which made the watchdog message disappear and reboot and halt a little quicker. 

Now I did the same with kernel 4.9.28 and I get a kernel panic when trying to reboot or halt.
Have compiled the watchdog back-in for now. 

Is there a way to get rid of this thing in kernel 4.9? like it was with 4.1 and 4.4?

paulvee, 2017-07-25:
Rodizio,

 

Your probably stumbling on why I raised the issue.

There are two watchdog methods.

The old one that has been around for ages is the watchdog package that needs to be installed.

 

However, with systemd, a "native" watchdog facility (not a package!) was introduced.

The two are incompatible with each other (the watchdog device is single user). You either use one or the other.

The mystery is how to properly use the systemd watchdog, which is why I raised the issue.

 

Also, there have been kernel updates that changed the behavior without notification (I could not find any).

 

Maybe this helps,

 

paulv