RPi-Distro/repo #76: plymouth-drm can't be installed

pozzugno, 2017-08-07:
> pi@raspberrypi:/etc$ **sudo apt-get install plymouth-drm**
> Reading package lists... Done
> Building dependency tree
> Reading state information... Done
> Some packages could not be installed. This may mean that you have
> requested an impossible situation or if you are using the unstable
> distribution that some required packages have not yet been created
> or been moved out of Incoming.
> The following information may help to resolve the situation:
> 
> The following packages have unmet dependencies:
>  plymouth-drm : Depends: plymouth (= 0.8.5.1-5) but it is not going to be installed
>                 Recommends: plymouth-themes-all
> E: Unable to correct problems, you have held broken packages.

> pi@raspberrypi:~$ **sudo apt-get install plymouth-drm plymouth**
> Reading package lists... Done
> Building dependency tree
> Reading state information... Done
> Some packages could not be installed. This may mean that you have
> requested an impossible situation or if you are using the unstable
> distribution that some required packages have not yet been created
> or been moved out of Incoming.
> The following information may help to resolve the situation:
> 
> The following packages have unmet dependencies:
>  plymouth : Breaks: plymouth-drm (< 0.9.0-6~) but 0.8.5.1-5 is to be installed
>  plymouth-drm : Depends: plymouth (= 0.8.5.1-5) but 0.9.2-3+rpi1 is to be installed
>                 Recommends: plymouth-themes-all
> E: Unable to correct problems, you have held broken packages.

XECDesign, 2017-08-07:
I don't see it in the jessie or stretch repos. 

What's the output of `apt-cache policy plymouth-drm`?

pozzugno, 2017-08-07:
> pi@raspberrypi:~$ **apt-cache policy plymouth-drm**
> plymouth-drm:
>   Installed: (none)
>   Candidate: 0.8.5.1-5
>   Version table:
>      0.8.5.1-5 0
>         500 http://archive.raspbian.org/raspbian/ wheezy/main armhf Packages

XECDesign, 2017-08-07:
Since wheezy is not supported, I highly recommend moving to Jessie or Stretch.

pozzugno, 2017-08-07:
I'm working with a [PiTFT](https://www.adafruit.com/product/2298) module. As suggested in the [tutorial](https://learn.adafruit.com/adafruit-pitft-28-inch-resistive-touchscreen-display-raspberry-pi), I downloaded and installed the ready-to-use distribution [2016-10-18-pitft-28r-lite.zip](https://s3.amazonaws.com/adafruit-raspberry-pi/2016-10-18-pitft-28r-lite.zip).

This file is described as "_Jessy-lite based PiTFT 2.4", 2.8" and 3.2" Resistive Image for Pi 1, 2, 3, Zero (Sept 23, 2016)_". I don't know why apt-get checks wheezy and not jessie.

Anyway the problem is the same. In jessie plymouth-drm doesn't exist, so I can't follow the [tutorial](https://github.com/notro/fbtft/wiki/Bootsplash) here that says to install the package plymouth-drm.

XECDesign, 2017-08-07:
Might be worth contacting adafruit to find out why the image uses the wheezy repo or notro to find out what the correct steps are on a stock jessie install.

pozzugno, 2017-08-07:
```
pi@raspberrypi:~$ cat /etc/apt/sources.list
deb http://mirrordirector.raspbian.org/raspbian/ jessie main contrib non-free rp
i
# Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://archive.raspbian.org/raspbian/ jessie main contrib non-free rpi
deb http://apt.adafruit.com/raspbian/ jessie main

I'm not an expert, I think apt-get tries to search packages on jessie sources, but fails for plymouth-drm, because it really doesn't exist on jessie. So it searches for previous distributions and found it on wheezy, with wrong dependencies.

I think the tutorial on notro is outdated. Maybe now the process to enable a splash screen on PiTFT early during the boot is different. Maybe plymouth-drm isn't necessary anymore.

XECDesign, 2017-08-07:
What's in `/etc/apt/sources.list.d/*` ?

pozzugno, 2017-08-07:
```
pi@raspberrypi:~$ ls /etc/apt/sources.list.d/*
/etc/apt/sources.list.d/raspi.list  /etc/apt/sources.list.d/wheezy.list
pi@raspberrypi:~$ cat /etc/apt/sources.list.d/raspi.list
deb http://archive.raspberrypi.org/debian/ jessie main ui
# Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://archive.raspberrypi.org/debian/ jessie main ui
pi@raspberrypi:~$ cat /etc/apt/sources.list.d/wheezy.list
deb http://archive.raspbian.org/raspbian wheezy main

pi@raspberrypi:~$
```

XECDesign, 2017-08-07:
Okay, they have added wheezy.list for some reason.

closed 2017-08-07 by XECDesign