RPi-Distro/repo #77: Please build a Kodi package for Stretch

andrum99, 2017-08-17:
The Kodi package in this (archive.raspberrypi.org) repo works well on Jessie. The RPF just released a Stretch-based distro, and unfortunately the version that is being pulled in from the Raspbian repo (not this repo) is a bit broken - it uses X, is rather slow, and the startup script is borked.

This is a feature request to have a new Kodi package in this repo for Stretch. The provision of a systemd service script would be nice as well - you should be able to use the one for the Raspbian package as a template.

LeSpocky, 2017-08-19:
I can confirm `kodi -fs --standalone` does not work anymore but crashes with "GUI not found". In the log it seems he tries to find an X server but can't. Starting from X is terribly slow on RPi2, taking several seconds for one keystroke to be reacted on. 😞

XECDesign, 2017-08-31:
Updated version of kodi from Rascas has been added to the repo.

closed 2017-08-31 by XECDesign

LeSpocky, 2017-08-31:
**Note:** upgrading from the previous kodi package fails. You have to remove it first and then install it again.

andrum99, 2017-08-31:
Is there a suitable systemd startup script?