RPi-Distro/repo #78: Unique local static IPv6 address issue on Rasbian Stretch

antoniosk, 2017-08-24:
Setting a static IPv4 configuration (address, mask, gateway) through dhcpcd.conf works perfectly. When I am trying to set a static IPv6 configuration on /etc/network/interfaces, the IPv6 configuration is not applied.

As a workaround I am doing an ifdown / ifup after restart. Restarting networking.service has no effect.

The very same configuration is working flawlessly in Debian/Rasbian Jessie and Debian Stretch, using the attached /etc/network/interfaces (with the corresponding adapter name). 

I installed Rasbian Stretch with Desktop (released on 2017-08-16) from scratch using the dd command.

[etc.network.interfaces.txt](https://github.com/RPi-Distro/repo/files/1249378/etc.network.interfaces.txt)

s1awa, 2017-09-11:
@antoniosk Read this https://raspberrypi.stackexchange.com/a/37921

> The Raspbian released in May 2015 changed the way networking (and particularly WiFi) works.
> Raspbian Stretch implements the same networking, however there are minor differences to configuration (mostly simplification) and **changed interface names**.
> 
> **This makes all the existing tutorials obsolete.**

XECDesign, 2017-09-11:
@antoniosk could you please post the output of `sudo journalctl`?

antoniosk, 2017-09-11:
In j1.txt you may find an (almost) complete log after system restart (I have filtered some records of no interest). Can't find anything suspicious here.

After issuing the commands sudo ifdown enxb827eb868c4b && sudo ifup enxb827eb868c4b the IPv6 address is registered normally without again any log records, except of the avahi-daemon as seen in j2.txt.
[j1.txt](https://github.com/RPi-Distro/repo/files/1293148/j1.txt)
[j2.txt](https://github.com/RPi-Distro/repo/files/1293149/j2.txt)

XECDesign, 2017-09-11:
What's the output of 'apt-cache policy dhcpcd5'

antoniosk, 2017-09-11:
dhcpcd5:
  Installed: 1:6.10.1-1+rpi3
  Candidate: 1:6.10.1-1+rpi3
  Version table:
 *** 1:6.10.1-1+rpi3 500
        500 http://archive.raspberrypi.org/debian stretch/main armhf Packages
        100 /var/lib/dpkg/status
     6.10.1-1 500
        500 http://mirrordirector.raspbian.org/raspbian stretch/main armhf Packages

XECDesign, 2017-09-11:
If you run `sudo systemctl disable dhcpcd` and reboot, does it behave as you expect?

antoniosk, 2017-09-11:
The interface enxb827eb868c4b remains down. If I run ifup it comes up normally with the IPv6 address configured in /etc/network/interfaces.

XECDesign, 2017-09-11:
Thanks, I'll dig in and see if I can reproduce the problem myself tomorrow.

antoniosk, 2017-09-11:
Well, it seems I found two solutions ( actually one with two parts):
Solution 1: I disabled dhcpcd putting both static IPv4 and IPv6 configuration in /etc/network/interfaces. The interface didn't come up automatically (as I said previously), so I added the line 
auto enxb827eb868c4b 
in /etc/network/interfaces and it worked!
Solution 2: I re - enabled dhcpcd, removed the static IPv4 configuration from /etc/network/interfaces (because the static IPv4 configuration is at /etc/dhcpcd.conf), but I left the line auto enxb827eb868c4b in place.

I can confirm, however, that the absence of "auto <interface name>" in Debian/Raspbian Jessie and Debian Stretch doesn't prevent static IPv6 configuration from working.

@XECDesign Thank you very much for your help!

XECDesign, 2017-09-12:
@antoniosk sorry, I should've noticed the missing 'auto'. That would explain it.

Why are you using dhcpcd.conf and interfaces at the same time? Why not use one or the other to configur ipv4 and ipv6? I ask because dhcpcd should refuse to run if you're using interfaces, so this is likely to stop working for you in the not too distant future.

antoniosk, 2017-09-13:
My static IPv4 configuration resides in dhcpcd.conf but this doesn't work for IPv6. I tried to make it work using information from the man pages and from documentation on Internet without luck. That's why the static IPv6 configuration is still in interfaces.

XECDesign, 2017-09-13:
Looked into it and you're quite right. Our version of dhcpcd just missed out on static ipv6 support. We're currently testing the latest version and it looks okay so far.

When that's release, it should all work through dhcpcd.conf with something like this:
```
interface eth0
		   noipv6rs
		   static ip_address=192.168.0.10/24
		   static ip6_address=fd51:42f8:caae:d92e::ff/64
		   static routers=192.168.0.1
		   static domain_name_servers=192.168.0.1  fd51:42f8:caae:d92e::1
```

antoniosk, 2017-09-13:
Yes, this is given as an example configuration on FreeBSD 11 dhcpcd.conf(5) man page. I am not running FreeBSD, but that should refer to dhcpcd version 6.11.5. Raspbian Stretch has dhcpcd version 6.10.1.

XECDesign, 2017-09-13:
Yup, we'll be updating to 6.11.5 if we don't find any issues with it.

s1awa, 2017-09-14:
I got an update for dhcpcd on Raspbian Stretch yesterday. Can't remember the version though.

XECDesign, 2017-09-22:
Heads up that 6.11.5 has been in the repo for a few days, so you should be able to set a static IPv6 IP without touching /etc/network/interfaces

antoniosk, 2017-09-22:
Dhcpcd updated today. I reverted /etc/network/interfaces to the original version and put all my static IPv4 and IPv6 configuration in /etc/dhcpcd.conf according to the man page.
Works perfectly now!

XECDesign, 2017-09-25:
Thanks for confirming

closed 2017-09-25 by XECDesign