RPi-Distro/repo #79: OpenGL performance reduced on Raspbian Stretch

runboy93, 2017-08-29:
Originally reported here:
https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=191791

Basically, OpenGL performance is reduced by Stretch update when compared to older Jessie release.

Jessie used Mesa 13.0.0 release, Stretch use Mesa 13.0.6 release.

quote from user:

"Running 'glxgears -info' on Raspbian Jessie is reporting around 180 frames-per-second rendering speed but the same on Stretch is reporting only about 45 FPS, or a quarter of the speed! This is on the same hardware (a Raspberry Pi 3 with HDMI output), in both cases with the 'experimental' VC4 GL Driver disabled."

# glxinfo output for Jessie and Stretch

Jessie:

name of display: :0.0
display: :0 screen: 0
direct rendering: Yes
server glx vendor string: SGI
server glx version string: 1.4
client glx vendor string: Mesa Project and SGI
client glx version string: 1.4
GLX version: 1.4
OpenGL vendor string: VMware, Inc.
OpenGL renderer string: Gallium 0.4 on llvmpipe (LLVM 3.9, 128 bits)
OpenGL core profile version string: 3.3 (Core Profile) Mesa 13.0.0
OpenGL core profile shading language version string: 3.30
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL version string: 3.0 Mesa 13.0.0
OpenGL shading language version string: 1.30
OpenGL context flags: (none)
OpenGL ES profile version string: OpenGL ES 3.0 Mesa 13.0.0
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.00

Stretch:

name of display: :0.0
display: :0 screen: 0
direct rendering: Yes
server glx vendor string: SGI
server glx version string: 1.4
client glx vendor string: Mesa Project and SGI
client glx version string: 1.4
GLX version: 1.4
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: VMware, Inc. (0xffffffff)
    Device: llvmpipe (LLVM 3.9, 128 bits) (0xffffffff)
    Version: 13.0.6
    Accelerated: no
    Video memory: 927MB
    Unified memory: no
    Preferred profile: core (0x1)
    Max core profile version: 3.3
    Max compat profile version: 3.0
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.0
OpenGL vendor string: VMware, Inc.
OpenGL renderer string: Gallium 0.4 on llvmpipe (LLVM 3.9, 128 bits)
OpenGL core profile version string: 3.3 (Core Profile) Mesa 13.0.6
OpenGL core profile shading language version string: 3.30
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL version string: 3.0 Mesa 13.0.6
OpenGL shading language version string: 1.30
OpenGL context flags: (none)
OpenGL ES profile version string: OpenGL ES 3.0 Mesa 13.0.6
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.00

Note: llvmpipe is being used on both Jessie and Stretch releases (not VC4)

CPU load:
Jessie: ~180 fps, CPU load ~50% (CPU meter in task bar) ~180% ('top' command), colours correct.
Stretch: ~45 fps, CPU load ~70% (CPU meter) ~260% ('top'), colours incorrect.

XECDesign, 2017-09-04:
After a bit of testing, it does look like this is an issue which affects software OpenGL rendering in general, not just glxgears. However, the change of behaviour comes from upstream Debian packages and software opengl rendering is not something we've ever touched or claimed to support. I would recommend enabling vc4 fkms in raspi-config for proper OpenGL support.

XECDesign, 2017-09-22:
Closing since it's an upstream problem that we can't do much about.

closed 2017-09-22 by XECDesign

katastic, 2018-04-24:
So your guy's solution is (between here and that thread)... Raspberry Pi is broken and we don't give a crap? A year later and no change.

I got this problem from a fresh install, in 2018, no changes to Pi. Glxgears is colored and games run at 1 FPS.

Going to raspi-config and setting "full KMS" fixes the color... and gives even less framerate.

I'm now the proud owner of two electronic paperweights.

XECDesign, 2018-04-24:
Your issue does not sound relevant to this at all. Out of the box, raspberry Pi supports OpenGL ES, not OpenGL. If you're running OpenGL games, out of the box, software rendering is used, so 1FPS sounds about right. Games which use OpenGL ES and have appropriately sized textures run great on the pi, but unfortunately there aren't many.

There's the experimental Mesa driver being developed by Broadcom which adds an open source driver and OpenGL support. That has some limitations and a long way to go, but basic games run great there as well. Oolite works without problems if you limit texture sizes. Neverball, many other games and applications just work without any special modifications with this driver. However, it is still experimental.

The relevant part of your post is the colour issue with glxgears, which seems to be the result of an issue in llvmpipe. AIUI, it's a problem with llvm's armv6 code generation. In general, you should report issues upstream to developers who are familiar with the code and are best able to address the issue. There are no LLVM or Mesa experts at Raspberry Pi. We can't fix every single upstream issue.

If you bought your Pi solely for the purpose of running OpenGL games, I would recommend checking supported APIs in advance.

kivutar, 2018-09-22:
I understand that there are no LLVM or Mesa experts in the RPF, but this issue is affecting the user experience in such a way that it will end up affecting the reputation of the Raspberry Pi itself.

I am maintaining a Linux distro for the RPi. And when an upstream bug is affecting my distro enough to make my users complain, we usually do this:

 * keep the issue open
 * tag it as Important
 * initiate communication with upstream
 * post a reference to the bug declared on upstream tracker
 * revert the download links on the public website to point to the last version that was working properly
 * not issue new release until the bug is fixed upstream

ab1jx, 2019-11-12:
This should be easy enough to test or fix by removing every deb you have installed that mentions OpenGL.  Software OpenGL is not what you want.  Or try it with a fresh image.  I'm usually careful to avoid the software emulation stuff but this machine's down around 50 FPS so I checked and sure enough, it's crapped up.

dpkg-query -l | grep -i opengl
ii  glmark2-es2-fbdev-odroid              20190710-1+deb10                    arm64        OpenGL ES 2.0 fbdev benchmark (ODROID build)
ii  libepoxy0:arm64                       1.5.3-0.1                           arm64        OpenGL function pointer management library
ii  libgl1-mesa-dri:arm64                 18.3.6-2                            arm64        free implementation of the OpenGL API -- DRI modules
ii  libglew2.1:arm64                      2.1.0-4                             arm64        OpenGL Extension Wrangler - runtime environment
ii  libglu1-mesa:arm64                    9.0.0-2.1+b3                        arm64        Mesa OpenGL utility library (GLU)
ii  libglx-mesa0:arm64                    18.3.6-2                            arm64        free implementation of the OpenGL API -- GLX vendor library
ii  libopengl0:arm64                      1.1.0-1                             arm64        Vendor neutral GL dispatch library -- OpenGL support
ii  libvdpau-va-gl1:arm64                 0.4.2-1+b1                          arm64        VDPAU driver with OpenGL/VAAPI backend
ii  mesa-utils-extra                      8.4.0-1+b1                          arm64        Miscellaneous Mesa utilies (opengles, egl)

For the VideoCore OpenGL ES most of that shouldn't be there.  Software emulation is much slower and it's eating about 70% of the CPU.  Stick with the /opt/vc stuff.

XECDesign, 2019-11-12:
> This should be easy enough to test or fix by removing every deb you have installed that mentions OpenGL.

The issues mentioned here have been resolved in Buster.

>  Software OpenGL is not what you want.

Sometimes it is, sometimes it's not. But Mesa provides both hardware and software opengl and opengl es.

From the output of your commands, it looks like you're running something other than Raspbian.

> For the VideoCore OpenGL ES most of that shouldn't be there. Software emulation is much slower and it's eating about 70% of the CPU. Stick with the /opt/vc stuff.

Again, yes those packages should be there if you want hardware acceleration on the pi 4. There's no VideoCore OpenGL ES for pi 4, so Mesa is the only choice.

ab1jx, 2020-02-22:
I have no Pi 4, don't plan to buy one.  I have (3x) RPI 3B, 1 each Pinebook Pro, Rock64, Odroid N2.
Partly what we're seeing is an effort to get rid of x11 and force us to use Wayland, or not, the x11 drivers just don't get written by the hardware manufacturers.  They provide one driver and it's Wayland.  Which leaves the world floundering and reverse engineering which might well take until Wayland is ready for mass consumption.  Mesa isn't always accelerated.  Vulkan is maybe a solution.

XECDesign, 2020-02-24:
> I have no Pi 4, don't plan to buy one. I have (3x) RPI 3B,

Then you have the choice of using the legacy GL libs or the more standard and fully open Mesa libs (which are also hardware accelerated).

> Partly what we're seeing is an effort to get rid of x11 and force us to use Wayland, or not, the x11 drivers just don't get written by the hardware manufacturers.

Mesa provide glamor and glx, which allows X11 to use hardware rendering without requiring a special X11-specific driver.

I'm not too familiar with how wayland works, but I expect it's a similar situation there. We're certainly not pushing anybody towards it.

> Mesa isn't always accelerated.

It is if you enable FKMS.

ab1jx, 2020-02-25:
Getting information out of the GPU designers is like pulling hen's
teeth, at least partly because they're afraid of their competitors
stealing their secrets.  So for Malis ARM has only released
closed-source binary blob drivers for Wayland.  For the right price
they might do x11 too.  But most of their money probably comes from
selling Malis in millions of Android phones.  And I now have as many
Malis as videocore.

I'm not so much interested in acceleration for the speed as I am in
being able to write shaders so the multiple processors the GPU can
take some of the load of rendering off the CPU.  And my app doesn't do
typical ray-tracing type rendering at all, it's more of an animated
plot with scrolling (downward).  So I'm looking into Vulkan, because
it's intended to be a low-level API.