RPi-Distro/repo #80: VC libraries missing errors with Raspbian Stretch

runboy93, 2017-08-29:
Reported here:
https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=191857
https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=191638

After Raspbian Stretch update, there are many errors shown on applications / games:

"error while loading shared libraries: libGLESv2.so: cannot open shared object file: No such file or directory"

"Starting game from /home/pi/littlecrane-rpi
./littlecrane: error while loading shared libraries: libEGL.so: cannot open shared object file: No such file or directory"

Original posts from users:

"I can't run a lot of games in new raspbian stretch, they worked OK in jessie:

Code: Select all

error while loading shared libraries: libGLESv2.so: cannot open shared object file: No such file or directory

I'm using a fresh instalation of latest debian stretch in a raspberry pi 2.
Some of the games than now doen't work are Maldita Castilla, Hurrican, Cubixmania, They need to be feed, etc..."

"When I updated to Stretch from Jessie the picam application on my RPi stopped working because it couldn't find the libraries mentioned above. I tried sym linking and making copies of the corresponding brcm libraries so no avail. ldconfig was not impressed :-)

By chance I decided to run sudo rpi-update this morning: I had updated the OS, so maybe I should update the firmware too. The missing libraries are now back in /opt/vc/lib. They have last modified dates of 22:49 on August 23rd. picam is now working again.

Posted in case there are other users and applications affected by the apparent loss of these libraries."

runboy93, 2017-08-29:
"Raspbian Stretch the library is called "libbrcmGLESv2.so" and not "libGLESv2.so" as in Jessie"

runboy93, 2017-08-30:
"By chance I decided to run sudo rpi-update this morning: I had updated the OS, so maybe I should update the firmware too. The missing libraries are now back in /opt/vc/lib. They have last modified dates of 22:49 on August 23rd. picam is now working again."

Most likely fixed on newer rpi-update firmware (4.9.44?)

XECDesign, 2017-08-31:
Duplicate of https://github.com/RPi-Distro/firmware/issues/3

closed 2017-08-31 by XECDesign