RPi-Distro/repo #81: dhcpcd: fails to reconfigure/restart NTP with DHCP-provided NTP servers

bwarden, 2017-08-31:
In ntp-4.2.8p10, /etc/init.d/ntp now looks for a DHCP-provided configuration file in /run/ntp.conf.dhcp. The hook in dhcpcd5-6.10.1-1, **/lib/dhcpcd/dhcpcd-hooks/50-ntp.conf**, still points to the older location, /var/lib/ntp/ntp.conf.dhcp.

Workaround: explicitly setting $NTP_DHCP_CONF to "/run/ntp.conf.dhcp"

With this workaround, /run/ntp.conf.dhcp is properly generated upon restarting the dhcpcd service, however, it does not restart the ntp service. Enabling debug on dhcpcd reveals that the hook is apparently looking for the ntpd service, not the ntp service:
`Aug 31 09:38:09 rocky dhcpcd[2609]: Failed to get unit file state for ntpd.service: No such file or directory`

Workaround: explicitly setting $ntp_service to "ntp"

XECDesign, 2017-09-11:
Thanks, @bwarden, that's a very helpful bug report.
Could you please give this version a go?
[dhcpcd5_6.10.1-1+rpi4_armhf.zip](https://github.com/RPi-Distro/repo/files/1292663/dhcpcd5_6.10.1-1.rpi4_armhf.zip)

It has the following patch applied:
```diff
--- a/dhcpcd-hooks/50-ntp.conf
+++ b/dhcpcd-hooks/50-ntp.conf
@@ -37,16 +37,16 @@
 if [ -z "$ntp_service" ]; then
        case "$NTP_CONF" in
        *chrony.conf)           ntp_service=chronyd;;
-       *)                      ntp_service=ntpd;;
+       *)                      ntp_service=ntp;;
        esac
 fi
 
 # Debian has a seperate file for DHCP config to avoid stamping on
 # the master.
-if [ "$ntp_service" = ntpd ] && type invoke-rc.d >/dev/null 2>&1; then
+if [ "$ntp_service" = ntp ] && type invoke-rc.d >/dev/null 2>&1; then
        [ -e /var/lib/ntp ] || mkdir /var/lib/ntp
        : ${ntp_service:=ntp}
-       : ${NTP_DHCP_CONF:=/var/lib/ntp/ntp.conf.dhcp}
+       : ${NTP_DHCP_CONF:=/run/ntp.conf.dhcp}
 fi
 
 : ${ntp_restart_cmd:=service_condcommand $ntp_service restart}
```

bwarden, 2017-09-11:
Works as expected, thanks!

XECDesign, 2017-09-11:
Thanks. It should show up in the repo tomorrow.

closed 2017-09-12 by XECDesign