RPi-Distro/repo #82: adduser does not create mailbox file automatically on Raspbian Stretch lite

antoniosk, 2017-09-04:
It seems that the option CREATE_MAIL_SPOOL=yes in /etc/default/useradd is ignored. The expected behaviour is to create a file /var/mail/<username>. This is how it works in Raspbian Jessie.

Workaround: create the mailbox manually and set the permissions as follows (filename and username are the same):
chown <username>:mail <filename>
chmod 600 <filename>

Raspbian Stretch lite was installed using dd command with image released on 2017-08-16.

XECDesign, 2017-09-04:
Any idea what the behaviour in regular Debian Stretch is?

antoniosk, 2017-09-04:
Debian Stretch has the same problem.

XECDesign, 2017-09-04:
This repo is for issues that originate in the raspberrypi.org apt repo, so this should probably reported here instead:
https://www.debian.org/Bugs/Reporting

antoniosk, 2017-09-04:
Will do, thank you.

closed 2017-09-04 by XECDesign