RPi-Distro/repo #84: chromium-browser: missing dependency

puck, 2017-09-05:
Hey,

The chromium-browser in Raspbian Stretch is missing a package dependency on libgtk-3-0.

Error display when running chromium-browser from the command line:
/usr/lib/chromium-browser/chromium-browser: error while loading shared libraries: libgtk-3.so.0: cannot open shared object file: No such file or directory

Cheers,
Andrew

XECDesign, 2017-09-05:
Thanks. This is probably not enough to warrant a rebuild, but it should be added for the next build.

@kusti8 (who kindly provides us with the .deb's), could this be added?

runboy93, 2017-09-05:
chromium-browser_60.0.3112.113-0ubuntu0.14.04.1194_armhf.deb

kusti8, 2017-09-05:
Done!

XECDesign, 2017-09-05:
That was quick.

Thanks!

puck, 2017-09-05:
Awesome!

theHilikus, 2017-10-19:
Any idea when this fixed package will be in the repos?

wturnerharris, 2017-11-12:
Is there a merge request or something for this? I just had to manually install libgtk-3-0 manually. Not that it's terribly hard to do, but it would be helpful to simply install.

XECDesign, 2018-01-02:
Is the issue resolved?

kusti8, 2018-01-02:
A new version hasn't come out yet, so I haven't rebuilt it. I can build it if you need it.

XECDesign, 2018-01-02:
I've manually adjusted the existing .deb because there was another issue with it. AFAIK, the version that's currently in the repo should have this issue addressed.

closed 2018-01-30 by XECDesign