RPi-Distro/repo #85: gstreamer1.0-omx

mistralol, 2017-09-09:
Looks like raspbian stretch has gstreamer 1.10.4. Yet the gstreamer1.0 package has not been updated in a long time which is still at version 1.0.1 It is also really buggy.

Running pipelines works on the git repo for omx but not on the installed pipelines. eg things like

gst-launch-1.0 rtspsrc ! h264parse ! omxh264dec ! omxh264enc ! fakesink and various combinations crash on the current version. Another example would be "rtspsrc ! rtph264depay ! h264parse ! omxh264dec ! videoscale ! video/x-raw, width=320, height=200 ! omxh264enc ! fakesink silent=false"

Both of which run on the later version.

XECDesign, 2017-09-11:
I am not sure I follow.
```
libgstreamer1.0-0:
  Installed: 1.10.4-1
```
Where are you getting 1.0.1 from?

mistralol, 2017-09-11:
Sorry completly wrong package name in my report I have accidently ommited "-omx".

gstreamer1.0 is 1.10.4 however gstreamer-omx is 1.0.x which is badly out of date and doesn't work in many situations. "apt-cache show gstreamer-omx" will show the details.

Package: gstreamer1.0-omx
Source: gst-omx1.0
Version: **1.0.0.1-0+rpi12+jessiepmg**
Architecture: armhf
Maintainer: Andrew Lee (李健秋) <andrew.lee@collabora.co.uk>
Installed-Size: 254
Depends: libc6 (>= 2.4), libglib2.0-0 (>= 2.37.3), libgstreamer-plugins-base1.0-0 (>= 1.0.0), libgstreamer1.0-0 (>= 1.0.0)
Homepage: http://gstreamer.freedesktop.org/
Priority: optional
Section: rpi/libs
Filename: pool/rpi/g/gst-omx1.0/gstreamer1.0-omx_1.0.0.1-0+rpi12+jessiepmg_armhf.deb
Size: 85492

These details are also showen in the output of "gst-inspect-1.0 omxh264enc"

Plugin Details:
  Name                     omx
  Description              GStreamer OpenMAX Plug-ins
  Filename                 /usr/lib/arm-linux-gnueabihf/gstreamer-1.0/libgstomx.so
  Version                  **1.0.0.1**
  License                  LGPL
  Source module            gst-omx
  Source release date      2015-04-18 00:42 (UTC)
  Binary package           GStreamer OpenMAX Plug-ins git
  Origin URL               Unknown package origin

However compiling the offical gstreamer-omx package from the gstreamer releases for 1.10.4 seems to resolve many of the problems I am seeing.

XECDesign, 2017-09-11:
That makes more sense. Thanks.

If you already have the git version compiled and installed, would you be able to confirm whether it works correctly with Epiphany browser?

mistralol, 2017-09-11:
It looks like it works. Not exactly sure how to test it. Tried it on a youtube video but it seemed somewhat slow. Not sure how it behaved under the old version.

XECDesign, 2017-09-14:
I am not having much luck building 1.10.4:
```
./configure --enable-maintainer-mode --enable-gtk-doc --with-omx-target=rpi --prefix=/usr
```
```
make[2]: Entering directory '/home/pi/dev/gst-omx/omx'
  CC       libgstomx_la-gstomx.lo
  CC       libgstomx_la-gstomxbufferpool.lo
  CC       libgstomx_la-gstomxvideo.lo
  CC       libgstomx_la-gstomxvideodec.lo
In file included from /usr/include/gstreamer-1.0/gst/gl/egl/gstglmemoryegl.h:32:0,
                 from gstomxvideodec.c:41:
/usr/include/gstreamer-1.0/gst/gl/egl/gstglcontext_egl.h:66:37: error: unknown type name ‘EGLAttrib’
       EGLClientBuffer buffer, const EGLAttrib *attrib_list);
                                     ^~~~~~~~~
Makefile:716: recipe for target 'libgstomx_la-gstomxvideodec.lo' failed
make[2]: *** [libgstomx_la-gstomxvideodec.lo] Error 1
make[2]: Leaving directory '/home/pi/dev/gst-omx/omx'
Makefile:517: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/home/pi/dev/gst-omx'
Makefile:449: recipe for target 'all' failed
make: *** [all] Error 2

```

XECDesign, 2017-09-15:
It's alright, I've got debs built from a modified gst-omx package. Just doing a bit more testing and looking into why the existing package is there (since it comes from raspbian.org and not debian or raspberrypi.org).

I will upload them for testing in a bit.

XECDesign, 2017-09-21:
The updates package should in available in the repo now. Please let me know if it solves or introduces any issues.

closed 2017-09-22 by XECDesign

ihora, 2018-01-31:
Dear all,

I am trying to recompile gstreamer omx plugin since I am experiencing this problem while playing stream with codecs compiled this way:
./autogen.sh --with-omx-header-path=/opt/vc/include/IL --with-omx-target=rpi

gstomxvideodec.c:974:gst_omx_video_dec_reconfigure_output_port:<omxh264dec-omxh264dec0>
Failed to negotiate RGBA for EGLImage 

According to:
http://gstreamer-devel.966125.n4.nabble.com/omxh264dec-Failed-to-negotiate-RGBA-for-EGLImage-td4685740.html#a4685749
http://gstreamer-bugs.narkive.com/e7uPa917/bug-767801-new-failed-to-negotiate-rgba-for-eglimage-on-raspberry-pi-3
and https://jp.raspberrypi.org/forums/viewtopic.php?f=66&t=193152

So when exporting these variables:
export CFLAGS="-I/opt/vc/include -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux/ -I/opt/vc/include/IL"
export LDFLAGS="-L/opt/vc/lib"
export CPPFLAGS="-I/opt/vc/include -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -I/opt/vc/include/IL"
export LIBS="-L/opt/vc/lib"

I ended up having this error:
/usr/include/gstreamer-1.0/gst/gl/egl/gstglcontext_egl.h:66:37: error: unknown type name ‘EGLAttrib’
       EGLClientBuffer buffer, const EGLAttrib *attrib_list);

Does anyone know resolution for this?
Thank you very much
Best regards,

XECDesign, 2018-01-31:
@ihora This repo is for reporting issues with packages. For general support, please visit the forum.

ihora, 2018-01-31:
I amm sorry, I would like to use the package, but the package gstreamer1.0-omx-rpi  yields the same problem therefore I am building it from sources.

ihora, 2018-01-31:
Ok, I would only kindly ask you whether the video hardware acceleration works in current Raspberry Pi gst-omx package.
Thank you

ihora, 2018-01-31:
Or is there any way how to properly install the package as:
sudo apt-get install gstreamer1.0-omx-rpi-config, installs packages, but gst-inspect-1.0 does not yield any support, I have to mangle other OMX packages in order to get it in gst-inspect-1.0 output.

XECDesign, 2018-01-31:
My understanding is that decoding works, but there's a bug in encoding. Taking a closer look at gst-omx is on the todo list, but it will be a while before I can get to it.

ihora, 2018-02-01:
Can I help somehow? Is there a tutorial somewhere how to build the package? We use encoding well on a camera with no performance issues, but we face problems using decoding.

XECDesign, 2018-02-01:
I think you fill get most support on the forum.
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=200306&start=25#p1260276