RPi-Distro/repo #86: networking and dhcpcd are both enabled by default

s1awa, 2017-09-11:
After installing Rasbian Stretch and updating to the latest Stretch version
`sudo apt dist-upgrade`
Both networking and dhcpcd services are running by default.
This messes with custom network configuration.

After disabling networking
`sudo systemctl disable networking`
my dhcpcd configuration for a static IP finally started working

XECDesign, 2017-09-11:
dhcpcd would refuse to run if anything is configured in /etc/network/interfaces.

s1awa, 2017-09-12:
nothing is in /etc/network/interfaces

XECDesign, 2017-09-12:
So networking wouldn't do anything.

s1awa, 2017-09-14:
Yes networking should not do anything. But it prevented dhcpcd from setting static IP on boot until I disabled networking.
I am using my PI as dhcp server with pihole on my home network and don't have too much free time.
So I can't really mess around with the networking anymore to get some logs.

Hopefully the new dhcpcd version will solve those issues.

closed 2017-09-14 by s1awa